project-chip/connectedhomeip #2495: Android APP does not work because of missing libCHIPController.so

KalmanWang, 2020-09-08:
**Problem**

It seems that there is no description about how to generate the library libCHIPController.so, right?
And the following issue occurs when I try to run the android APP:

> java.lang.UnsatisfiedLinkError: dalvik.system.PathClassLoader[DexPathList[[zip file "/data/app/com.google.chip.chiptool-naCOgNejJLwWRwlIyi4AVA==/base.apk"],nativeLibraryDirectories=[/data/app/com.google.chip.chiptool-naCOgNejJLwWRwlIyi4AVA==/lib/x86, /system/lib, /system/product/lib]]] couldn't find "libCHIPController.so

Could anyone tell me how to build the libCHIPController.so?

wgtdkp, 2020-09-08:
hi @KalmanWang  You need to follow the instructions at https://github.com/project-chip/connectedhomeip/blob/master/src/android/CHIPTool/README.md to build libraries required by the Android App.

KalmanWang, 2020-09-08:
> hi @KalmanWang You need to follow the instructions at https://github.com/project-chip/connectedhomeip/blob/master/src/android/CHIPTool/README.md to build libraries required by the Android App.

Hi @wgtdkp , I have finished all the steps but there is no file named "libCHIPController.so".
It seems the source code should be under ./src/controller/ ? However, I don't find any description about how to build the controller.

wgtdkp, 2020-09-08:
> Hi @wgtdkp , I have finished all the steps but there is no file named "libCHIPController.so".
> It seems the source code should be under ./src/controller/ ? However, I don't find any description about [...]

I have ran into this situation once on Linux. Removing the build directory and restart a clean build works for me.

KalmanWang, 2020-09-08:
> > > hi @KalmanWang You need to follow the instructions at https://github.com/project-chip/connectedhomeip/blob/master/src/android/CHIPTool/README.md to build libraries required by the Android App.
> > 
> > 
> > Hi @wgtdkp , I have finished all the steps but there is no file named "libCHIPController.so".
> > It seems the source code should be under ./src/controller/ ? However, I don't find any description about how to build the controller.
> 
> I have ran into this situation once on Linux. Removing the build directory and restart a clean build works for me.
> 
> > It seems the source code should be under ./src/controller/ ?
> 
> The library should be in the build directory, you should be able to find all libraries with
> 
> ```shell
> find ${BUILD_DIR} -name "*.so"
> ```
> 
> The same for `JAR` libraries.

Thanks, I will have a try.

KalmanWang, 2020-09-08:
@wgtdkp  It works. Thanks very much!

closed 2020-09-08 by KalmanWang