project-chip/connectedhomeip #2507: Add implementation of silab emberAfGetCurrentTime

shana-apple, 2020-09-08:
emberAfGetCurrentTime implementation is missing.

bzbarsky-apple, 2020-09-08:
The impl lives in `third_party/silabs/framework/util/time-util.c` in the Silicon Labs PR, which is one of the files we did not import.  That said, the whole impl is:
```
uint32_t emberAfGetCurrentTime(void)
{
#ifdef EMBER_AF_PLUGIN_TIME_SERVER
  return emAfTimeClusterServerGetCurrentTime();
#else
  return emberAfGetCurrentTimeCallback();
#endif
}
```
`emberAfGetCurrentTimeCallback` is expected to be implemented by the application, as far as I can tell.

`emAfTimeClusterServerGetCurrentTime` lives in `third_party/silabs/framework/plugin/time-server/time-server.c` and also tries calling `emberAfGetCurrentTimeCallback`.  If that fails, it looks for an endpoint with the server side of the Time cluster (0x000a) implemented on it and reads the "Time" attribute if it finds one.

bzbarsky-apple, 2023-03-23:
`emberAfGetCurrentTime` is gone entirely now.

closed 2023-03-23 by bzbarsky-apple