project-chip/connectedhomeip #2512: Flash script built inside Docker doesn't work outside Docker

szatmz, 2020-09-08:
#### Problem
Images (e.g. nrf5 examples) built inside docker cannot be flashed likely because the flash script does not have access to USB.

 #### Proposed Solution
Open USB up in docker image(s).

kpschoedel, 2020-09-08:
I don't know whether anything can be done about Docker permissions, but the flash script can be revised so that it works outside docker. The path names in it are currently absolute (i.e. with respect to the Docker container) so they make no sense outside. Changing them to be relative to the output directory and/or flash script (currently the same) would make it usable outside the container, and would also make sense for a CI configuration that might have building and flashing on different machines.

kpschoedel, 2020-09-09:
Looked at Docker a bit and it seems there is no good way to flash from _inside_ the container (portable, reliable, secure… pick at most one) though an individual developer may find `--device` or `--device-cgroup-rule` options work well enough for their case.

The current plan is for building to copy all files necessary for flashing into the output directory and remove any references to the build tree from the flashing scripts, so that flashing natively after building in a container works smoothly.

PR #2542 (2020-09-11): [GN] Factor out common code from platform *_firmware_utils.py
#### Problem

The two current platform scripts nrf5_firmware_utils.py and
efr32_firmware_utils.py have a great deal of code in common, which would
hamper maintenance and adding new platforms. Also, generated flashing
scripts contain absolute paths, which impedes Docker users and future CI.

#### Summary of Changes

* Move flashing scripts to //scripts/flashing/
* Slightly revise tool-calling code to make more of it shareable.
* Factored out common code into firmware_utils.py
* Builds copy all required files to the output directory.
 [... cut]
commit 3d90cf84 (2020-09-11): [GN] Factor out common code from platform *_firmware_utils.py (#2542)

* [GN] Factor out common code from platform *_firmware_utils.py

#### Problem

The two current platform scripts nrf5_firmware_utils.py and
efr32_firmware_utils.py have a great deal of code in common, which would
hamper maintenance and adding new platforms. Also, generated flashing
scripts contain absolute paths, which impedes Docker users and future CI.

#### Summary of Changes

* Move flashing scripts to //scripts/flashing/
* Slightly revise tool-calling code to make more of it shareable.
* Factored out common code into fi [... cut]
closed 2020-09-11 by mspang