project-chip/connectedhomeip #2529: Unable to build Android CHIP Tool

doru91, 2020-09-09:
I am trying to build the CHIP Tool for Android. I followed all the REAMDME steps (https://github.com/project-chip/connectedhomeip/blob/master/src/android/CHIPTool/README.md) and was able to generate and copy in the right locations the SetupPayloadParser.jar/libSetupPayloadParser.so.

When I try to build the Project in Android Studio I get an undefined reference to "devicecontroller" (see the image below). I noticed that the corresponding .h/.a files are located under src/controller but it's unclear to me how to link these .h/.a files to the project. Could you please help? 

[image]

@randyrossi

nileshvora441, 2020-09-09:
@doru91 I believe you will have to add **CHIPController.jar** and **libCHIPController.so** from **src\controller\java** directory just like you added **SetupPayloadParser.jar** from **src\setup_payload\java**.

Android CHIP Tool ReadMe must need to improve for the same.

doru91, 2020-09-09:
Thanks, @nileshvora441 , I just did it. But now it complains that about a missing io.openthread package. 

[image]

nileshvora441, 2020-09-09:
> Thanks, @nileshvora441 , I just did it. But now it complains that about a missing io.openthread package.
> 
> [image]

That's because you probably didn't follow step#11 from readme.
That step will also add some jar files and .so related to your error.

doru91, 2020-09-09:
Yep, I realized after I posted the message, I am working on building the jars for the ot commissioner. Thanks a lot!

nileshvora-sls, 2020-09-09:
It is obvious and It is confusing for reader., Step#11 must be Step#10.

doru91, 2020-09-09:
Right, I will send a PR with some README fixes. But I still got compilation errors after running Step 11 (which generates and copies the correct .so and .jar files).

[image]

doru91, 2020-09-09:
@nileshvora-sls , were you able to compile CHIP Tool on your setup?

nileshvora-sls, 2020-09-10:
> @nileshvora-sls , were you able to compile CHIP Tool on your setup?
> cc @andreilitvin

CHIP-tool is different example which supposed to be run from command line as of now AFAIK. and Android Project is different project. 
However I am able to compile both by just following steps. I am using Windows 10 subsystem for Linux (Ubuntu 18.04). I have installed NDK in Linux subsystem and And Installed latest Android Studio on Windows.

doru91, 2020-09-14:
I redo all the step and I was able to compile CHIP Tool. However, if I push the "Light On/Off Cluster" button the application crashes. Do you have the same problem? I am thinking that I should "Scan a QR code" first.

andy31415, 2020-09-14:
@vidhis88 may have more info.

I just came back from OOO ... 2 weeks ago I was able to compile and run though. To run, you do need an android device rather than emulator since BLE is required.

nileshvora-sls, 2020-09-15:
> I redo all the step and I was able to compile CHIP Tool. However, if I push the "Light On/Off Cluster" button the application crashes. Do you have the same problem? I am thinking that I should "Scan a QR code" first.

I didn't test whole APP. Initially after installing it, I came to know that it is in development stage. I though it would do provisioning over BLE for my ESP32. But only few independent modules are added in app.

vidhis88, 2020-09-16:
Provisioning over BLE is WIP for Android demo app.

About the crash from the Light ON/OFF cluster commands screen, it would be helpful to get the logs to diagnose.
Since provisioning is not in, currently, you need to get the CHIP device on to your Wi-Fi network and use the assigned IP as input in the light ON/OFF screen. Maybe you are already doing that but just double-checking.

doru91, 2020-09-18:
Here is the stack trace for the moment when I press the ZCL On-Off button:

2020-09-18 04:15:41.123 18925-18925/com.google.chip.chiptool E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.google.chip.chiptool, PID: 18925
    java.lang.UnsatisfiedLinkError: dalvik.system.PathClassLoader[DexPathList[[zip file "/data/app/com.google.chip.chiptool-4NzO8xGeYXACupyXcwxt4A==/base.apk"],nativeLibraryDirectories=[/data/app/com.google.chip.chiptool-4NzO8xGeYXACupyXcwxt4A==/lib/arm64, /system/lib64, /system/vendor/lib64]]] couldn't find "libCHIPController.so"
        at java.lang.Runtime.loadLibrary0(Runtime.java:1012)
        at java.lang.System.loadLibrary(System.java:1669)
        at chip.devicecontroller.ChipDeviceController.<clinit>(ChipDeviceController.java:81)
        at com.google.chip.chiptool.ChipClient.getDeviceController(ChipClient.kt:29)
        at com.google.chip.chiptool.clusterclient.OnOffClientFragment.getDeviceController(OnOffClientFragment.kt:18)
        at com.google.chip.chiptool.clusterclient.OnOffClientFragment.onCreateView(OnOffClientFragment.kt:28)
        at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2698)
        at androidx.fragment.app.FragmentStateManager.createView(FragmentStateManager.java:320)
        at androidx.fragment.app.FragmentManager.moveToState(FragmentManager.java:1187)
        at androidx.fragment.app.FragmentManager.moveToState(FragmentManager.java:1356)
        at androidx.fragment.app.FragmentManager.moveFragmentToExpectedState(FragmentManager.java:1434)
        at androidx.fragment.app.FragmentManager.moveToState(FragmentManager.java:1497)
        at androidx.fragment.app.BackStackRecord.executeOps(BackStackRecord.java:447)
        at androidx.fragment.app.FragmentManager.executeOps(FragmentManager.java:2169)
        at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:1992)
        at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:1947)
        at androidx.fragment.app.FragmentManager.execPendingActions(FragmentManager.java:1849)
        at androidx.fragment.app.FragmentManager$4.run(FragmentManager.java:413)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7078)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:964)
2020-09-18 04:15:41.142 18925-18925/com.google.chip.chiptool I/Process: Sending signal. PID: 18925 SIG: 9

vidhis88, 2020-09-18:
That seems to be the case -- you want to have the libCHIPController.so in src/main/jniLibs/arm64_v8a/ (for Android phones) and src/main/jniLibs/x86/ (for Android emulator).

doru91, 2020-09-19:
Thanks, once I fixed the location of libCHIPController.so the application works. 

One question though: was the ZCL functionality tested using IPv6 addresses? I am trying to connect to an IPV6 only device but CHIP Tool doesn't seem to generate any packet when I'm using IPv6 addresses. 

I have the following setup: CHIP Tool (running on Android) -> Border Router (15.4 and Wi-Fi connectivity) -> Thread device (15.4 connectivity).  Basically the CHIP Tool is connected to the Border Router using Wi-Fi while the Thread device is connected to the Border Router using 15.4 connectivity. My end-goal is to send ZCL on/off commands to the Thread device and the Border Router is used as a proxy. If I open the CHIP Tool, press the "Light On/Off Cluster" button then enter the IPv6 address of the Thread device, then no IPV6 packet is sent. I assume that this is a CHIP Tool problem because I can send ping packets between the phone and Thread device.

@vidhis88

wgtdkp, 2020-09-20:
@doru91 
I have done the same tests, and I was able to send CHIP commands to a Thread Device proxied by a Thread Border Router.

>  If I open the CHIP Tool, press the "Light On/Off Cluster" button then enter the IPv6 address of the Thread device then no IPV6 packet is sent.

How did you verified that there is no IPv6 packets sent? Did you capture traffic on the Border Router device? If not, I would suggest tp firstly make sure the path from the phone to the Border Router is good.

I would be helpful if you can share your setup and testing commands.

doru91, 2020-09-21:
@wgtdkp , I restarted the Android Phone and I was able to see IPv6 packets generated by the CHIP Tool. Maybe it was an internal Android routing issue...

One question: what IPV6 did you use on the Thread device for pinging? I tried to use a ULA IPv6 address but there is a CHIP rule that blocks registering of an IPV6 ULA on the LWIP interface:  https://github.com/project-chip/connectedhomeip/blob/master/src/platform/OpenThread/GenericThreadStackManagerImpl_OpenThread_LwIP.ipp#L182

The comment from the link above mentions that ULA addresses should be registered in the Warm module but I don't see this module inside Project CHIP.

khushbushah2302, 2020-09-22:
> Thanks, @nileshvora441 , I just did it. But now it complains that about a missing io.openthread package.
> 
> [image]

I am facing same error.
In step 11, `ABI=arm64-v8a API=21 ./third_party/ot-commissioner/build-android-libs.sh` this command gives below error.
`./build/src/common/libcommissioner-common.so: No such file or directory`
How to resolve this ? any idea ?

nileshvora-sls, 2020-09-23:
> I am facing same error.
> In step 11, `ABI=arm64-v8a API=21 ./third_party/ot-commissioner/build-android-libs.sh` this command gives below error.
> `./build/src/common/libcommissioner-common.so: No such [...]

Yes. Actually It was working previously but in recent restyle commit, script got changed and error is coming now. Modify build-android-libs.sh with following change. I just removed quotation mark at line number 42.

> for lib in $(find ./build -name "*.so"); do
>     ## Avoid copying symblink files.
>     [ ! -L "$lib" ] && cp "$lib" libs
> done

wgtdkp, 2020-09-25:
> @wgtdkp , I restarted the Android Phone and I was able to see IPv6 packets generated by the CHIP Tool. Maybe it was an internal Android routing issue...
> 
> One question: what IPV6 did you use on the Thread device for pinging? I tried to use a ULA IPv6 address but there is a CHIP rule that blocks registering of an IPV6 ULA on the LWIP interface: https://github.com/project-chip/connectedhomeip/blob/master/src/platform/OpenThread/GenericThreadStackManagerImpl_OpenThread_LwIP.ipp#L182
> 
> The comment from the link above mentions that ULA addresses should be registered in the Warm module but I don't see it inside Project CHIP.
> 
> cc @bukepo

@doru91 Sorry for delayed response. I don't think a ULA address can work in this case, I was using a GUA address. To add GUA address to Thread device, you need to execute below command on the Thread Border Router:
1. add an external prefix: `prefix add 2020:1993::/64 paros`.
2. update Network Data: `netdata register`.

You should be able see a GUA address started with `2020:1993` on both Thread Border Router and Thread end device, and you can now ping this address.

closed 2020-09-25 by bzbarsky-apple

Pavankm3, 2021-12-27:
Hello strucked in Chiptool which was build in linux machine and developing currently in android studio and facing issue when QR code scanned for provisioning device.
[image]