project-chip/connectedhomeip #2551: Remove an unnecessary null pointer check

elfring, 2020-09-10:
[An extra null pointer check is not needed](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") in [the function “ChipDeviceController::Shutdown”](https://github.com/project-chip/connectedhomeip/blob/6665e76f731de87ee3a290dcfb1e920fb0b08a0d/src/controller/CHIPDeviceController.cpp#L136 "Update candidate").

woody-apple, 2020-09-21:
Personally: Not sure what an "unnecessarily null check is" ;)

elfring, 2020-09-22:
How do you think about to [use a development tool like “clang-tidy” for corresponding source code adjustments](https://clang.llvm.org/extra/clang-tidy/checks/readability-delete-null-pointer.html "Deletion of unnecessary null pointer checks")?

closed 2021-09-29 by stale[bot]