project-chip/connectedhomeip #2569: BLEManagerImpl may have MTU overflow its storage

bzbarsky-apple, 2020-09-11:
#### Problem
In `src/platform/ESP32/BLEManagerImpl.cpp` we have this line in `BLEManagerImpl::HandleGATTCommEvent`:
```
conState->MTU = param->mtu.mtu;
```
The right-hand side is a `uint16_t`.  The left-hand side is a 10-bit bitfield.  There seems to be no check that the value on the right will fit on the left, nor any explanation for why a check is not needed.

 #### Proposed Solution
Either introduce a check or add an explanation for why it's not needed.

@sagar-apple it looks like you might know something about this code?

sagar-apple, 2020-09-11:
@shana-apple amd @vivien-apple are probably more familiar with this. I can look if they're unable.

bzbarsky-apple, 2020-09-11:
`src/platform/EFR32/BLEManagerImpl.cpp` has a similar issue in `BLEManagerImpl::UpdateMtu`.

andy31415, 2022-01-31:
This looks like platform specific code - re-assigned to platform owner.

dhrishi, 2022-02-01:
>nor any explanation for why a check is not needed.

@bzbarsky-apple As per the BLE specification, the maximum MTU value can be 517 bytes. This can be accomodated in 10 bits. Good enough to close this?

bzbarsky-apple, 2022-02-01:
Good enough, but would be even better with a comment.  ;)

PR #14693 (2022-02-03): esp32: Make SoftwareVersion configurable from application
#### Problem
- No way to configure software version from application
- Fixes #2569 

#### Change overview
- Added an config option to configure software version. This value will be read on reading basic cluster software version attribute.
- Added a comment explaining why the MTU won't overflow

#### Testing
Read basic cluster software version using chip-tool
closed 2022-02-03 by andy31415