project-chip/connectedhomeip #2608: Create automated functional tests for on/off cluster

bzbarsky-apple, 2020-09-14:
#### Problem
We have no automated testing for the on/off cluster.

 #### Proposed Solution
Create some.  That will need both infrastructure (e.g. cirque?) and actual testcase creation.

We may be able to use Silicon Labs tests as inspiration for the latter, in terms of what needs to be tested.

bzbarsky-apple, 2020-09-14:
@erjiaqing

erjiaqing, 2020-09-17:
What is the expected behavior of such test? For now, the cirque test will test the linux mock device which uses on-off cluster.

bzbarsky-apple, 2020-09-17:
Things that I think are worth testing, with verifications of expected states via Read Attributes after every step:

1) Send the 'off' command to endpoints 1 and 2 (might need to add support for endpoint 2 to the linux mock device).
1) Send an 'on' command to endpoint 1.
1) Send a 'toggle' command to endpoint 2.
1) Send an 'off' command to endpoint 2.
1) Send an 'on' command to endpoint 2.
1) Send a 'toggle' command to endpoint 1.
1) Send a 'toggle' command to endpoint 1.
1) Send an 'off' command to endpoint 1.

That seems like a bare-minimum set of things we should test.  Again, we should consider looking at the Silicon Labs tests to see what they are testing.

It would also be good if we could check that the server side is getting the attribute changed callbacks with the right values...

closed 2021-11-04 by stale[bot]