project-chip/connectedhomeip #2660: Unable to test chip-tool example with ESP32 BLE rendezvous

nileshvora-sls, 2020-09-15:
**Problem**
ESP32 is up with BLE rendezvous and I am trying to run chip-tool example on Ubuntu Machine which has Bluetooth. But getting following error.

> root@nilesh:~/Desktop/chip/connectedhomeip/examples/chip-tool/build# ./chip-tool.out echo-ble 3840 012345678
> CHIP:DL: CHIP task running
> CHIP:DL: CHIP:TOO: Failed to connect to the device
> CHIP:TOO: Run command failure: Error 4003 (0x00000FA3)
> Timer: 2592000000
> Usage:
> ./chip-tool.out command [params]
> 
> Supported commands and their parameters:
> echo-ble discriminator setup-pin-code
> echo device-remote-ip device-remote-port
> on device-remote-ip device-remote-port endpoint-id
> off device-remote-ip device-remote-port endpoint-id
> toggle device-remote-ip device-remote-port endpoint-id
> read device-remote-ip device-remote-port endpoint-id attr-name
> 
> Supported attribute names for the 'read' command:
> onoff
> 

**Proposed Solution**
Provide or update readme to test chip-tool with ESP32 or any other supported hardware.

vivien-apple, 2020-11-02:
At the moment Rendezvous is not supported on **chip-tool**. Assigning to myself since I was planning to do it at some point.

woody-apple, 2021-01-25:
SW BRB: Any update @vivien-apple ?

vivien-apple, 2021-02-02:
I guess this has been closed by #4347

closed 2021-02-02 by vivien-apple