project-chip/connectedhomeip #2680: android: remove use of shared libc++

mspang, 2020-09-16:
Middleware should not use shared libc++ as this can cause compatibility issues. We should instead link libc++ statically (our use of libc++ should be really minimal anyway).

- [ ] Hide all dynamic symbols except JNI_OnLoad and use RegisterNatives to hook up native code to JNI
- [ ] Built with -static-libstdc++

mspang, 2020-09-16:
@vidhis88

andy31415, 2020-10-23:
NDK generally states 'only use static libc++ if you have exactly one shared library' which is not currently the case for CHIP.

Trying with 'none' if we use no C++ except new/delete could be an option if we really use no C++. Unsure if we can in any way hide all dynamic symbols. Especially since new/delete would be public and NDK describes allocating in one so and freeing in another as being one of the potential dangers.

closed 2021-10-30 by stale[bot]