project-chip/connectedhomeip #2712: ChipCryptoTests are failing

mspang, 2020-09-18:
ChipCryptoTests are failing with both automake & GN

```
====================================================
   Chip ge3775af: src/crypto/tests/test-suite.log
====================================================

# TOTAL: 1
# PASS:  0
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

FAIL: qemu_crypto_tests.sh
==========================

esptool.py v2.9-dev
1+0 records in
1+0 records out
4194304 bytes (4.2 MB, 4.0 MiB) copied, 0.0134144 s, 313 MB/s
6+1 records in
6+1 records out
24608 bytes (25 kB, 24 KiB) copied, 6.82e-05 s, 361 MB/s
0+1 records in
0+1 records out
3072 bytes (3.1 kB, 3.0 KiB) copied, 3.34e-05 s, 92.0 MB/s
17+1 records in
17+1 records out
1162992 bytes (1.2 MB, 1.1 MiB) copied, 0.000654602 s, 1.8 GB/s
Adding SPI flash device
==60125==WARNING: ASan doesn't fully support makecontext/swapcontext functions and may produce false positives in some cases!
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6908
load:0x40078000,len:13720
load:0x40080400,len:3880
entry 0x40080680
[0;32mI (496) boot: ESP-IDF v4.1 2nd stage bootloader[0m
[0;32mI (513) boot: compile time 20:51:03[0m
[0;32mI (532) boot: chip revision: 0[0m
[0;32mI (548) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (551) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (553) boot.esp32: SPI Flash Size : 2MB[0m
[0;32mI (579) boot: Enabling RNG early entropy source...[0m
[0;32mI (622) boot: Partition Table:[0m
[0;32mI (624) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (629) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (635) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (639) boot:  2 factory          factory app      00 00 00010000 001f0000[0m
[0;32mI (660) boot: End of partition table[0m
[0;32mI (680) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3aa98 (240280) map[0m
[0;32mI (1188) esp_image: segment 1: paddr=0x0004aac0 vaddr=0x3ffbdb60 size=0x04190 ( 16784) load[0m
[0;32mI (1256) esp_image: segment 2: paddr=0x0004ec58 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (1302) esp_image: segment 3: paddr=0x0004f060 vaddr=0x40080400 size=0x00fb8 (  4024) load[0m
[0;32mI (1351) esp_image: segment 4: paddr=0x00050020 vaddr=0x400d0020 size=0xc4388 (803720) map[0m
[0;32mI (2850) esp_image: segment 5: paddr=0x001143b0 vaddr=0x400813b8 size=0x17b14 ( 97044) load[0m
[0;32mI (3161) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (3165) boot: Disabling RNG early entropy source...[0m
[0;32mI (3196) cpu_start: Pro cpu up.[0m
[0;32mI (3198) cpu_start: Application information:[0m
[0;32mI (3201) cpu_start: Project name:     chip-tests[0m
[... 81 lines cut ...]
[0;31mE (33508) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:[0m
[0;31mE (33508) task_wdt:  - IDLE1 (CPU 1)[0m
[0;31mE (33508) task_wdt: Tasks currently running:[0m
[0;31mE (33508) task_wdt: CPU 0: IDLE0[0m
[0;31mE (33508) task_wdt: CPU 1: tester[0m
[ CHIP Crypto PAL tests : Test adding entropy sources                                ] : PASSED
[ CHIP Crypto PAL tests : Test PBKDF2 SHA256                                         ] : PASSED
[0;31mE (38508) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:[0m
[0;31mE (38508) task_wdt:  - IDLE1 (CPU 1)[0m
[0;31mE (38508) task_wdt: Tasks currently running:[0m
[0;31mE (38508) task_wdt: CPU 0: IDLE0[0m
[0;31mE (38508) task_wdt: CPU 1: tester[0m
[ CHIP Crypto PAL tests : Test P256 Keygen                                           ] : PASSED
***ERROR*** A stack overflow in task tester has been detected.
abort() was called at PC 0x4008da90 on core 1

ELF file SHA256: 766efcbb9d6de0a8

Backtrace: 0x4008d6e5:0x3ffd7e00 0x4008da79:0x3ffd7e20 0x4008da90:0x3ffd7e40 0x4008ef63:0x3ffd7e60 0x400908cc:0x3ffd7e80 0x40090882:0x3ffd7e80 0xffffffff:0x3ffd7e80 |<-CORRUPTED

Rebooting...
idf: *** ERROR:  Tests did not run to completion.
FAIL qemu_crypto_tests.sh (exit status: 1)

```

bzbarsky-apple, 2020-09-18:
@pan-apple

pan-apple, 2020-09-21:
This might be due to upgrading to a newer ESP IDF version + adding new features in Crypto APIs (which in turn depend on additional mbedTLS APIs). The failure seems to be related to stack overflow. We can try increasing the stack size for QEMU configuration.

PR #4987 (2021-02-24): Increase stack size for esp32 unit tests
#### Problem

Stack overflow running `TestCSR_Gen()` in `src/crypto/tests/CHIPCryptoPALTest.cpp`

#### Summary of Changes

Increase test task stack from 8K to 12K.

Fixes #2712 ChipCryptoTests are failing on esp32 qemu
commit cc5daa90 (2021-02-24): Allocate more stack for esp32 tests (#4987)

#### Problem

Stack overflow running `TestCSR_Gen()`
in `src/crypto/tests/CHIPCryptoPALTest.cpp`

#### Summary of Changes

Increase test task stack from 8K to 12K.

Fixes #2712 ChipCryptoTests are failing on esp32 qemu
closed 2021-02-24 by mspang