project-chip/connectedhomeip #2721: TestTCP fails on Mac, prevents build completion

bzbarsky-apple, 2020-09-18:
#### Problem
After #2689 landed, when I run `./gn_build.sh` I get a log like so:

```
'#0:','Test-CHIP-Tcp'
'#2:','Setup                 ','PASSED'
'#3:','Simple Init Test IPV4 ','PASSED'
CHIP node ready to service events; PID: 13071; PPID: 13066
Connection closed.Connection complete encountered an error: OS Error 2006 (0x000007D6): Device not configuredSO_LINGER: 22Connection closed.Freeing closed connection.Failed to initialize TCP transport: OS Error 2048 (0x00000800): Address already in useFailed to initialize TCP transport: OS Error 2048 (0x00000800): Address already in useAsync DNS worker thread woke up.Async DNS worker thread exiting.Async DNS worker thread woke up.Async DNS worker thread exiting.../../src/transport/tests/TestTCP.cpp:138: assertion failed: "ReceiveHandlerCallCount == 1"
'#3:','Message Self Test IPV4','FAILED'
../../src/transport/tests/TestTCP.cpp:85: assertion failed: "err == CHIP_NO_ERROR"
'#3:','Simple Init Test IPV6 ','FAILED'
../../src/transport/tests/TestTCP.cpp:117: assertion failed: "err == CHIP_NO_ERROR"
../../src/transport/tests/TestTCP.cpp:135: assertion failed: "err == CHIP_NO_ERROR"
../../src/transport/tests/TestTCP.cpp:138: assertion failed: "ReceiveHandlerCallCount == 1"
'#3:','Message Self Test IPV6','FAILED'
'#4:','Teardown              ','PASSED'
'#6:','3','4'
'#7:','5','8'
INF Test 1/1: [FAIL] TestTCP
[36/58] ACTION //src/transport/tests:TestTCP_run(//build/toolchain/host:mac_x64_gcc)
FAILED: mac_x64_gcc/gen/src/transport/tests/TestTCP_run.pw_pystamp 
python ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root /Users/bzbarsky/connectedhomeip/ --current-path /Users/bzbarsky/connectedhomeip/src/transport/tests --default-toolchain=//third_party/pigweed/repo/pw_toolchain/dummy:dummy --current-toolchain=//build/toolchain/host:mac_x64_gcc --touch /Users/bzbarsky/connectedhomeip/out/debug/mac_x64_gcc/gen/src/transport/tests/TestTCP_run.pw_pystamp --capture-output -- /Users/bzbarsky/connectedhomeip/third_party/pigweed/repo/pw_unit_test/py/pw_unit_test/test_runner.py --runner ../../third_party/pigweed/repo/targets/host/run_test --test mac_x64_gcc/tests/TestTCP
INF Test 1/1: [ RUN] TestTCP
ERR ../../third_party/pigweed/repo/targets/host/run_test exited with status 252
OUT [13083]
Failed to initialize TCP transport: OS Error 2048 (0x00000800): Address already in useFailed to initialize TCP transport: OS Error 2048 (0x00000800): Address already in use'#0:','Test-CHIP-Tcp'
'#2:','Setup                 ','PASSED'
../../src/transport/tests/TestTCP.cpp:85: assertion failed: "err == CHIP_NO_ERROR"
'#3:','Simple Init Test IPV4 ','FAILED'
../../src/transport/tests/TestTCP.cpp:117: assertion failed: "err == CHIP_NO_ERROR"
../../src/transport/tests/TestTCP.cpp:135: assertion failed: "err == CHIP_NO_ERROR"
CHIP node ready to service events; PID: 13089; PPID: 13083
Failed to initialize TCP transport: OS Error 2048 (0x00000800): Address already in useFailed to initialize TCP transport: OS Error 2048 (0x00000800): Address already in useAsync DNS worker thread woke up.Async DNS worker thread exiting.Async DNS worker thread woke up.Async DNS worker thread exiting.../../src/transport/tests/TestTCP.cpp:138: assertion failed: "ReceiveHandlerCallCount == 1"
'#3:','Message Self Test IPV4','FAILED'
../../src/transport/tests/TestTCP.cpp:85: assertion failed: "err == CHIP_NO_ERROR"
'#3:','Simple Init Test IPV6 ','FAILED'
../../src/transport/tests/TestTCP.cpp:117: assertion failed: "err == CHIP_NO_ERROR"
../../src/transport/tests/TestTCP.cpp:135: assertion failed: "err == CHIP_NO_ERROR"
../../src/transport/tests/TestTCP.cpp:138: assertion failed: "ReceiveHandlerCallCount == 1"
'#3:','Message Self Test IPV6','FAILED'
'#4:','Teardown              ','PASSED'
'#6:','4','4'
'#7:','8','8'
INF Test 1/1: [FAIL] TestTCP
ninja: build stopped: subcommand failed.
```

 #### Proposed Solution
Fix the test to actually work.  And probably add newlines to some of that logging that is obviously missing them....

@andy31415

andy31415, 2020-09-21:
I got address in use on linux in some cases as well - apparently Inet::Close leaves sockes in SO_WAIT (if using Abort we are setting SO_LINGER to 0). I found no API available to close it immediately.

Someone who understands the TCPEndPoint may need to take a look as well ... I am unclear if this is an enpoint limitation or if I am using it incorrectly,

woody-apple, 2021-01-25:
SW BRB: @bzbarsky-apple is this still broken?

bzbarsky-apple, 2021-01-27:
Yes, this is still broken.  `src/transport/raw/tests/BUILD.gn` has:
```
  if (current_os != "mac") {
    test_sources += [ "TestTCP.cpp" ]
  }
```
and if I remove the condition and run the test unconditionally it fails on Mac with pretty much the same output as in the original filing of this issue.

turon, 2022-01-31:
TCP should be tested and pass on darwin for v1.
Removing `v1_triage_split_4`.

PR #15197 (2022-02-15): [mac] Turn on TestTCP
#### Problem

`TestTCP` is disabled on Mac. While working on #15138 I end up fixing it to validate that it works.
It was failing because the code that fetch the pending bytes uses `TIOCOUTQ` which is not the way to go on Mac.

fix #2721

#### Change overview
 * Update `TCPEndPointImpSockets.cpp` to fetch the pending bytes properly on Mac
 * Activate `TestTCP` on Mac

#### Testing
`TestTCP` is running.