project-chip/connectedhomeip #2722: gn_build.sh runs tests twice

bzbarsky-apple, 2020-09-18:
#### Problem
`gn_build.sh` seems to run tests twice: once as part of the `all` target and once as part of the `check` target.

 #### Proposed Solution
Remove the `check` from `time ninja -C "$CHIP_ROOT/out/debug" all check`

@turon @mspang

mspang, 2020-09-18:
It runs the tests once with gcc/mbedtls, once with gcc/openssl, and once with clang/openssl. This matches the presubmit testing for Linux.

See the docs under "unified build" for a discussion of this.

If you just do
  source scripts/activate.sh
  gn gen out/default
  ninja -C out/default all check

the tests will run once, with the default compiler & openssl.

mspang, 2020-09-18:
BTW I'm happy to discuss whether we need `gn_build.sh` going forward or what exactly it should do.

In my mind is has been semi-deprecated from the start, and the documentation does not even mention it in favor of documenting the underlying commands.

OTOH, it's helpful because it runs the whole sync, build, & test process and takes the user largely out of the equation which removes opportunities for error. But I also don't want anyone thinking this is the only way to build.

bzbarsky-apple, 2020-09-19:
That's fair.  Stepping back to actual use cases, what I appreciate about `gn_build.sh` is that it lets me compile a pretty wide swath of the tree with a single command if I set the right env vars.  That's been pretty useful in terms of shortening the edit/compile cycle time when working on "treewide" changes of various sorts; it's a lot faster than pushing to a GitHub branch and waiting for CI.  I'd be even happier if it also built the esp32 client; I've been doing that manually in parallel with running `gn_build.sh`.

For my purposes so far, having it run the unit tests has not been particularly beneficial, but I definitely see value in having a way to do "build and run unit tests" as well....

mspang, 2020-09-21:
If you set up the args the same way as `gn_build.sh` you do get separate testing. Compilation is all in the `default` target:

`ninja -C out/debug default   # or just leave targets blank`

Testing is in the "check" target

`ninja -C out/debug check

ninja `all` builds every defined target by definition so it does both. I see there is an inconsistency here because `all` builds and tests everything but e.g. `all_host_gcc` only builds - maybe we should call that `build_host_gcc` instead.

That `gn_build.sh` it doesn't do exactly what you'd prefer is exactly the problem I see with `gn_build.sh` vs underlying commands - it is inflexible. But as I said inflexibility can be a strength in some circumstances, so I'm on the fence about it. Of course we can add options to `gn_build.sh`, but if you set things up and run ninja directly you have full control over what targets to execute.

I'd ideally like to simplify the SDK management so hopefully even without `gn_build.sh` it would still not be hard to set up a unified build. In fact it could be even easier, with automating fetching of SDKs into third_party.

I agree it would be ideal to get esp32 & zephyr into the same set up.

woody-apple, 2021-01-25:
SW BRB: @mspang What do you think the next steps are here? Can we clarify the remaining items?

mspang, 2021-01-26:
I think the original problem statement in not accurate in suggesting than identical work is duplicated, so we should probably just close it.

With everything on this script builds 11 different ways and runs tests 3 different ways but I find this rather valuable :). If anyone would prefer it to do something else let's file another bug and discuss it there.

closed 2021-01-26 by mspang