project-chip/connectedhomeip #2727: nRF5 lighting app does not work after #2010

erjiaqing, 2020-09-21:
#### Problem
```
git checkout c9025fee
(cd examples/lighting-app/nrf5 && rm -rf out && gn gen out/debug && ninja -C out/debug && ./out/debug/chip-nrf52840-lighting-example.flash.py)
# not work
git checkout 7aa6f539
(cd examples/lighting-app/nrf5 && rm -rf out && gn gen out/debug && ninja -C out/debug && ./out/debug/chip-nrf52840-lighting-example.flash.py)
# works
```

 #### Proposed Solution
<suggested fix, suggested enhancement>

erjiaqing, 2020-09-21:
I tried to use GDB but get no results.

```
  % JLinkRTTClient
###RTT Client: ************************************************************ 
###RTT Client: *               SEGGER Microcontroller GmbH                * 
###RTT Client: *   Solutions for real time microcontroller applications   * 
###RTT Client: ************************************************************ 
###RTT Client: *                                                          * 
###RTT Client: *       (c) 2012 - 2016  SEGGER Microcontroller GmbH       * 
###RTT Client: *                                                          * 
###RTT Client: *     www.segger.com     Support: support@segger.com       * 
###RTT Client: *                                                          * 
###RTT Client: ************************************************************ 
###RTT Client: *                                                          * 
###RTT Client: * SEGGER J-Link RTT Client   Compiled Jan 17 2020 17:29:00 * 
###RTT Client: *                                                          * 
###RTT Client: ************************************************************ 

###RTT Client: -----------------------------------------------
###RTT Client: Connecting to J-Link RTT Server via localhost:19021 ...
###RTT Client: Connected.

SEGGER J-Link V6.60e - Real time terminal output
J-Link OB-SAM3U128-V2-NordicSemi compiled Jan 21 2020 17:30:48 V1.0, SN=683385043
Process: JLinkExe
[00000000] <info> app: ==================================================
[00000000] <info> app: chip-nrf52840-lighting-example starting
[00000000] <info> app: *** DEVELOPMENT BUILD ***
[00000000] <info> app: ==================================================
[00000000] <info> app: Enabling SoftDevice
[00000000] <info> app: Waiting for SoftDevice to be enabled
[00000000] <info> app: SoftDevice enable complete
[00000000] <info> app: SoftDevice BLE enabled
[00000000] <info> app: Init CHIP stack
[00000000] <info> chip: Starting BLE initialization
[00000000] <info> chip: [DL] BLEManagerImpl::Init() complete
[00000000] <info> app: Initializing OpenThread stack
[00000000] <info> chip: [DL] Setting OpenThread device type to ROUTER
[00000000] <info> app: Starting CHIP task
[00000000] <info> app: Starting OpenThread task
[00000000] <info> app: System Heap Utilization: heap size 40960, arena size 25988, in use 25976, free 12
[00000000] <info> app: Starting FreeRTOS scheduler
[00000000] <debug> chip: [DL] Thread task running
[00000000] <info> chip: [IN] New pairing for key 0!!
```

@oleggladyshev @Damian-Nordic  any ideas?

Damian-Nordic, 2020-09-21:
Hi,

First of all, note that you may use nRF Connect version of the lighting-app. nRF Connect makes common development tasks a bit easier by providing useful commands for them. For instance, it takes only "west debug" command to set up a gdb session.

Anyway, if you want to use the legacy nRF5 SDK, you should use JLinkGDBServer to attach to the device and then attach to the gdb server with gdb for arm. For example, run something like that:

JLinkGDBServer -If SWD -Speed 4000 -Device nRF52
arm-non-eabi-gdb
> target remote localhost:2331

2331 is the default port for JLinkGDBServer.

Let me know if that helps,
Damian

PR #2820 (2020-09-25): [nRF5 examples] Make lighting/lock app work again
 #### Problem
After #2010 is merged, the lighting app does not work.

 #### Summary of Changes
Uncomment the code that is commented out in #2010 

#### Testing
build the image, flash the image on nrf52840dk, verify that chip-tool can control the device.

fixes #2727
closed 2020-09-25 by rwalker-apple