project-chip/connectedhomeip #2732: GN build system requires internet to build

bhaskar-apple, 2020-09-21:
#### Problem
GN build system requires internet to build and also seems to rely on Google servers. 

Here is an output snippet when one builds the ESP32 without internet connection
```
[P86481 17:06:17.578 client.go:248 W] RPC failed transiently. Will retry in 8m32s {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/DescribeInstance\": dial tcp: i/o timeout", "host":"chrome-infra-packages.appspot.com", "method":"DescribeInstance", "service":"cipd.Repository", "sleepTime":"8m32s"}
```

mspang, 2020-09-21:
Building doesn't access the network. Bootstrap does access the internet (it always has as deps are scattered around the net).

Currently GN & ninja binaries are downloaded from Pigweed's infra for major platforms. The bootstrap also uses PyPI.

mspang, 2020-09-21:
You might be noticing that, starting from a clean tree, `gn_build.sh` will transparently bootstrap.

closed 2021-09-29 by stale[bot]