project-chip/connectedhomeip #2740: unsafe deletion in ChipDeviceController::Shutdown when !CONFIG_DEVICE_LAYER

mspang, 2020-09-22:
#### Problem
ChipDeviceController::Shutdown borrows a pointer to SystemLayer, InetLayer in the Android build:

https://github.com/project-chip/connectedhomeip/blob/46b482a8ba3fbef09a59f5e0c44e53f220d0200e/src/controller/CHIPDeviceController.cpp#L96-L97

This is deleted in Shutdown() here:

https://github.com/project-chip/connectedhomeip/blob/46b482a8ba3fbef09a59f5e0c44e53f220d0200e/src/controller/CHIPDeviceController.cpp#L120-L121

This looks unsafe as these are static objects in the Android build, which is currently the main user of this path:

https://github.com/project-chip/connectedhomeip/blob/46b482a8ba3fbef09a59f5e0c44e53f220d0200e/src/controller/java/CHIPDeviceController-JNI.cpp#L84-L85

 #### Proposed Solution

These deletions probably need to removed, even if it wasn't statically allocated it looks like a case of multiple ownership which would cause a double free.

mspang, 2020-09-22:
@vidhis88

msandstedt, 2021-05-27:
I would argue that in fact this should not occur in any of the builds.  Even if the controller is not passed SystemLayer and InetLayer, it gets references to these from DeviceLayer.  I don't see how in any cases DeviceController owns these.  The controller is not a singleton.  How can it ever be the proper owner of other singleton entities?

closed 2022-01-25 by woody-apple