project-chip/connectedhomeip #2748: All Embedded Device-side Code Needs to Be Scrubbed of Direct Heap Usage

gerickson, 2020-09-22:
#### Problem
The software produced by Project CHIP is consumed both inside and outside Project CHIP, across a variety of platforms. The capabilities of these platforms are broad, spanning soft real-time, deeply-embedded systems based on RTOSes that might cover life safety and/or physical security applications to richer, softly-embedded systems based on non-RTOS platforms such as Linux or Darwin. While the latter are apt to have fully-functional heaps, the former explicitly do not.

As a result, direct calls to malloc/free/new/delete and usage of C Standard Library, C++ Standard Library, C++ Standard Template Library (STL), or other library functionality that, in turn, call malloc/free/new/delete should be avoided and, ideally, eliminated.

#### Proposed Solution
1. Places where malloc/free/new/delete are explicitly invoked in embedded device-side code should be instead indirected through src/lib/support/CHIPMem.h, giving the stack configurator the option of how they want memory managed in their device.
2. Places where C++ STL are used should be replaced with a common singly- or doubly-linked list, implemented in src/lib/support or should a static bitmap or list allocator such as the attached, [nldatastructures.zip](https://github.com/project-chip/connectedhomeip/files/5262597/nldatastructures.zip).
3. Other patterns should be triaged and replacements designed with the appropriate indirections or other non-heap usage.

/cc/ @tcarmelveilleux, @woody-apple, @szatmz

woody-apple, 2020-09-22:
I think once https://github.com/project-chip/connectedhomeip/issues/2736 is in place, we should measure and baseline on that in respect to code size and heap. 

@andy31415 do we have a good way to test heap post a test run?

kpschoedel, 2020-09-24:
I'm starting to work on this. I'll make and mention explicit sub-issues so that it's clear what work is in progress and what is available.

kpschoedel, 2020-10-09:
I think the following may be the remaining C library calls (not covering C++ libraries yet) that may use heap under the hood. (It's not actually obvious to me what code is meant to run on a device, but if it's not obvious to me I don't think it will be obvious to a device maker either.)

## src/app/clusters/ias-zone-client/ias-zone-client.c
### Calls
    fp = fopen("iaszone.txt", "w");
    fp = fopen("iaszone.txt", "r");
    fscanf(fp, "%x ", &data1);
    fscanf(fp, "%x %x %x %x ", &data2, &data3, &data4, &data5);
    fscanf(fp, "%x ", &data1);
### Build Condition
    #if defined(EZSP_HOST) && !defined(EMBER_TEST) && defined(UNIX_HOST)
### Notes
- Some implementations of stdio may call malloc for the default buffer.
- Some implementations of the scanf have a GNU extension `m` modifier that calls malloc().

## src/inet/AsyncDNSResolverSockets.cpp
### Calls
    gaiReturnCode = getaddrinfo(resolver.asyncHostNameBuf, nullptr, &gaiHints, &gaiResults);
### Build Condition
    #if CHIP_SYSTEM_CONFIG_USE_SOCKETS && CHIP_INET_CONFIG_ENABLE_ASYNC_DNS_SOCKETS CHIP_INET_CONFIG_ENABLE_DNS_RESOLVER

## src/inet/DNSResolver.cpp
### Calls
    gaiReturnCode = getaddrinfo(hostNameBuf, nullptr, &gaiHints, &gaiResults);
### Build Condition
    #if CHIP_SYSTEM_CONFIG_USE_SOCKETS

## src/inet/InetInterface.cpp
### Calls
    int res = getifaddrs(&mAddrsList);
### Build Condition
    #if CHIP_SYSTEM_CONFIG_USE_SOCKETS && CHIP_SYSTEM_CONFIG_USE_BSD_IFADDRS

## src/inet/InetLayer.cpp
### Calls
    rv = getifaddrs(&ifaddr);
    freeifaddrs(ifaddr);
### Build Condition
    #if CHIP_SYSTEM_CONFIG_USE_SOCKETS && CHIP_SYSTEM_CONFIG_USE_BSD_IFADDRS

## src/inet/tests/TapAddrAutoconf.cpp
### Calls
    const int err = getifaddrs(&addrsList);
    tmp = (char *) malloc(strlen(buf));
    freeifaddrs(addrsList);
### Build Condition
    #if CHIP_SYSTEM_CONFIG_USE_SOCKETS && CHIP_SYSTEM_CONFIG_USE_BSD_IFADDRS

gerickson, 2020-10-12:
> I think the following may be the remaining C library calls (not covering C++ libraries yet) that may use heap under the hood. (It's not actually obvious to me what code is meant to run on a device, [...]

All of those would largely be on UNIX/POSIX/Linux/Darwin systems supporting a heap and malloc, so I don't think we have to deal with these.

kpschoedel, 2020-10-13:
Once I've got a similar list of config conditions for C++ library use, I intend to add a build-time check that they're combined with CHIP_CONFIG_MEMORY_MGMT_MALLOC.

kpschoedel, 2020-10-13:
For C++ `new` with constructors, is the preference for …?

1. placement-new into CHIPMem allocations
2. global :new calling CHIPMem

kpschoedel, 2020-10-19:
For the record, I'm currently looking through C++ library uses in the code to pick the next most important case to replace.

M1cha, 2021-07-07:
> For C++ `new` with constructors, is the preference for …?
> 
> 1. placement-new into CHIPMem allocations
> 2. global :new calling CHIPMem

IMO both are unacceptable because memory should be allocated statically by the linker and provided by the project-specific initialization code. (Basically like the current HomeKit ADK does it).
CHIPMem only changes the allocator function but doesn't eliminate the allocations.

Reasons for that:
- there might not be a memory area which can be used be multiple libraries which defeats the purpose of allocations. Take a look at the Zephyr RTOS. The heap is disabled by default and things like bluetooth or the IP stack have usecase-specific private memory pools with a fixed data-type and number of elements
- your app can never fail because it runs out of memory
- you can see the max RAM usage at compile-time

cpfister, 2021-09-17:
Merely indirecting malloc etc. calls to a facade like CHIPMem.h as described above does not solve the problem, I'm afraid. As described above, that facade would still provide a general dynamic memory management API to the programmers.

In contrast, the HomeKit ADK protocol code implementation was designed to _never_ allocate memory itself. It is sandwiched between a platform abstraction layer (PAL) below the protocol code, and the application above the protocol code. Depending on the platform, a PAL implementation may use malloc *internally*, and the application may pass memory to the protocol code that it has allocated using malloc. Makes sense for Linux-class devices with "infinitely large" memories.

However, this design allows resource-constrained PALs and apps for small microcontrollers to completely do away with dynamic memory management / a heap. For typical microcontrollers, both PAL and app statically pre-allocate memory in global variables as needed. As @M1cha said, this guarantees to never run into OOM conditions, which could otherwise easily happen with very small heaps. Works very well for HomeKit over a broad spectrum of microcontrollers.

M1cha, 2022-09-16:
ping because it's still an unresolved show-stopper

closed 2022-10-03 by stale[bot]

M1cha, 2022-10-03:
why did the stale bot close this? my post should've kept it active