project-chip/connectedhomeip #2757: Figure out what parts of libstdc++ we are pulling in and why

bzbarsky-apple, 2020-09-22:
#### Problem
Our esp32 "wifi-echo" example app has 114KB of flash used by code that comes from `libstdc++`.  I looked a bit at the specific functions we're pulling in and I see things like:
```
void std::__facet_shims::__numpunct_fill_cache<wchar_t>(std::integral_constant<bool, true>, std::locale::facet const*, std::__numpunct_cache<wchar_t>*)
std::collate<wchar_t>::do_compare(wchar_t const*, wchar_t const*, wchar_t const*, wchar_t const*) const
std::locale::_Impl::_Impl(unsigned int)
```
etc, etc.  So for some reason we're pulling in a bunch of locale stuff, and I strongly suspect this is not in fact desired.  Just that `locale::_Impl` constructor is 1.5KB of codesize!

 #### Proposed Solution
Figure out which parts of this we actually need and why and make the rest go away.  This might need to end up as multiple issues as we go....

bzbarsky-apple, 2020-09-22:
Oh, and we also need to check whether this is happening on other platforms too.  But I just looked at `out/debug/nrf5_lighting_app//chip-nrf52840-lighting-example.map` and it seems to have the locale bits too....

mspang, 2020-09-22:
https://github.com/project-chip/connectedhomeip/pull/2759

bzbarsky-apple, 2020-09-22:
> #2759

That helps a lot!  With that change, the `libstdc++` bits are down to more like 15KB.

bzbarsky-apple, 2021-10-29:
Still need to get to this.

tcarmelveilleux, 2022-01-31:
This impacts Flash and RAM size on all embedded platforms

kpschoedel, 2022-02-02:
The [ALLOW list](https://github.com/project-chip/connectedhomeip/pull/14704/files#diff-23cd8eee1a1c2914651abc53b8ed8061cf68b992f1232764462882decd02042eR93) in #14704 is a partial answer:

```

```