project-chip/connectedhomeip #2758: gn_build.sh is failing

vidhis88, 2020-09-22:
#### Problem
Running gn_build.sh fails with the following error log:
```
ninja: Entering directory `./out/debug'
[9/360] ar libCHIP.a
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: standalone/lib/libCHIP.a(lwip.dhcp6.c.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: standalone/lib/libCHIP.a(lwip.inet6.c.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: standalone/lib/libCHIP.a(lwip.api_lib.c.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: standalone/lib/libCHIP.a(lwip.api_msg.c.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: standalone/lib/libCHIP.a(lwip.if.c.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: standalone/lib/libCHIP.a(lwip.netbuf.c.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: standalone/lib/libCHIP.a(lwip.netdb.c.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: standalone/lib/libCHIP.a(lwip.netifapi.c.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: standalone/lib/libCHIP.a(lwip.sockets.c.o) has no symbols
[154/360] ACTION //src/transport/tests:TestUDP_run(//build/toolchain/host:mac_x64_gcc)
FAILED: mac_x64_gcc/gen/src/transport/tests/TestUDP_run.pw_pystamp 
python ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root /Users/vidhishah/Dev/CHIP/connectedhomeip/ --current-path /Users/vidhishah/Dev/CHIP/connectedhomeip/src/transport/tests --default-toolchain=//third_party/pigweed/repo/pw_toolchain/dummy:dummy --current-toolchain=//build/toolchain/host:mac_x64_gcc --touch /Users/vidhishah/Dev/CHIP/connectedhomeip/out/debug/mac_x64_gcc/gen/src/transport/tests/TestUDP_run.pw_pystamp --capture-output -- /Users/vidhishah/Dev/CHIP/connectedhomeip/third_party/pigweed/repo/pw_unit_test/py/pw_unit_test/test_runner.py --runner ../../third_party/pigweed/repo/targets/host/run_test --test mac_x64_gcc/tests/TestUDP
INF Test 1/1: [ RUN] TestUDP
ERR ../../third_party/pigweed/repo/targets/host/run_test exited with status 255
OUT [10879]
'#0:','Test-CHIP-Udp'
'#2:','Setup                 ','PASSED'
'#3:','Simple Init Test IPV4 ','PASSED'
CHIP node ready to service events; PID: 10880; PPID: 10879
Async DNS worker thread woke up.Async DNS worker thread exiting.Async DNS worker thread woke up.Async DNS worker thread exiting.../../src/transport/tests/TestUDP.cpp:136: assertion failed: "ReceiveHandlerCallCount == 1"
'#3:','Message Self Test IPV4','FAILED'
'#3:','Simple Init Test IPV6 ','PASSED'
'#3:','Message Self Test IPV6','PASSED'
'#4:','Teardown              ','PASSED'
'#6:','1','4'
'#7:','1','12'
INF Test 1/1: [FAIL] TestUDP
[169/360] ACTION //src/transport/tests:TestUDP_run(//config/mbedtls/toolchain:mac_x64_gcc_mbedtls)
FAILED: mac_x64_gcc_mbedtls/gen/src/transport/tests/TestUDP_run.pw_pystamp 
python ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root /Users/vidhishah/Dev/CHIP/connectedhomeip/ --current-path /Users/vidhishah/Dev/CHIP/connectedhomeip/src/transport/tests --default-toolchain=//third_party/pigweed/repo/pw_toolchain/dummy:dummy --current-toolchain=//config/mbedtls/toolchain:mac_x64_gcc_mbedtls --touch /Users/vidhishah/Dev/CHIP/connectedhomeip/out/debug/mac_x64_gcc_mbedtls/gen/src/transport/tests/TestUDP_run.pw_pystamp --capture-output -- /Users/vidhishah/Dev/CHIP/connectedhomeip/third_party/pigweed/repo/pw_unit_test/py/pw_unit_test/test_runner.py --runner ../../third_party/pigweed/repo/targets/host/run_test --test mac_x64_gcc_mbedtls/tests/TestUDP
INF Test 1/1: [ RUN] TestUDP
ERR ../../third_party/pigweed/repo/targets/host/run_test exited with status 255
OUT [10990]
'#0:','Test-CHIP-Udp'
'#2:','Setup                 ','PASSED'
'#3:','Simple Init Test IPV4 ','PASSED'
CHIP node ready to service events; PID: 10993; PPID: 10990
Async DNS worker thread woke up.Async DNS worker thread exiting.Async DNS worker thread woke up.Async DNS worker thread exiting.../../src/transport/tests/TestUDP.cpp:136: assertion failed: "ReceiveHandlerCallCount == 1"
'#3:','Message Self Test IPV4','FAILED'
'#3:','Simple Init Test IPV6 ','PASSED'
'#3:','Message Self Test IPV6','PASSED'
'#4:','Teardown              ','PASSED'
'#6:','1','4'
'#7:','1','12'
INF Test 1/1: [FAIL] TestUDP
ninja: build stopped: subcommand failed.
```

mspang, 2020-09-22:
I wasn't able to reproduce this failure.

vidhis88, 2020-09-22:
Restarting the terminal fixed the issue.

closed 2020-09-22 by vidhis88