project-chip/connectedhomeip #2762: Measure RAM usage of minimal "sensor" app

bzbarsky-apple, 2020-09-22:
#### Problem
We don't actually have a good idea of what RAM usage for a "minimal" CHIP app looks like.

 #### Proposed Solution
Take the simple "sensor" app that @shana-apple is building and measure its RAM usage.

We may want to do this multiple times, while having the app do whatever it knows how to do (worst-case, reply to Read Attributes commands), to see if we catch any fragmentation behavior going on.

gerickson, 2020-09-22:
I'd go one better on this. For a simple sensor app, there should be no heap usage and hence no runtime-dynamic RAM usage beyond what's defined by .bss and .data. That is, the RAM usage is known precisely at compile and link time. This is very much related to #2748.

bzbarsky-apple, 2020-09-23:
> no runtime-dynamic RAM usage beyond what's defined by .bss and .data.

Some bits of `.text` also go into RAM, as far as I can tell, and of course `.rel.text`.

> That is, the RAM usage is known precisely at compile and link time

That seems like a good bar to aim for, yes.

woody-apple, 2020-10-06:
Assigning to @shana-apple as she's going to lead this effort using the basic sensor

kpschoedel, 2020-10-28:
Looking at nrf5-lighting-app right now. Large static RAM allocation via .bss:
```
96672 total
├── 84154 third_party
│   └── 84154 connectedhomeip
│       ├── 76055 third_party
│       │   ├── 54956 openthread
│       │   │   └── 54956 repo
│       │   │       ├── 49704 src
│       │   │       │   └── 49452 core
│       │   │       │       └── 49452 common
│       │   │       │           └── 48456 instance.cpp
│       │   │       │               └── 48456 ot::gInstanceRaw
│       │   │       ├── 4050 third_party
│       │   │       │   └── 4050 NordicSemiconductor
│       │   │       │       └── 4050 drivers
│       │   │       │           └── 4050 radio
│       │   │       │               ├── 2064 nrf_802154_rx_buffer.c
│       │   │       │               │   └── 2064 nrf_802154_rx_buffers
│       │   │       │               └── 1433 mac_features
│       │   │       │                   └── 1371 ack_generator
│       │   │       │                       └── 1237 nrf_802154_ack_data.c
│       │   │       └── 1202 examples
│       │   │           └── 1202 platforms
│       │   │               └── 1202 nrf528xx
│       │   │                   └── 1202 src
│       │   ├── 16843 lwip
│       │   │   └── 16843 repo
│       │   │       └── 16843 lwip
│       │   │           └── 16843 src
│       │   │               └── 16815 core
│       │   │                   ├── 15024 memp.c
│       │   │                   │   └── 12131 memp_memory_PBUF_POOL_base
│       │   │                   └── 1635 mem.c
│       │   │                       └── 1619 ram_heap
│       │   └── 4256 jlink
│       │       └── 4256 segger_rtt
│       │           └── 4256 RTT
│       │               └── 4256 SEGGER_RTT.c
│       │                   └── 4096 _acUpBuffer
│       ├── 4948 examples
│       │   └── 4824 common
│       │       └── 4824 chip-app-server
│       │           └── 4824 Server.cpp
│       │               ├── 1784 (anonymous namespace)::gRendezvousServer
│       │               ├── 1672 (anonymous namespace)::gTestPairing
│       │               └── 1368 (anonymous namespace)::gSessions
│       └── 3151 src
├── 9899 [section .bss]
└── 2474 nrf5sdk
    └── 1959 components
        └── 1955 libraries
            └── 1377 log
                └── 1377 src
                    └── 1308 nrf_log_frontend.c
                        └── 1060 m_log_data
```

kpschoedel, 2020-10-28:
Looking at nrf5-lighting-app right now. Large flash allocation:

```
SECTION: .text
430856 total
├── 234718 third_party
│   └── 234718 connectedhomeip
│       ├── 181946 third_party
│       │   ├── 161294 openthread
│       │   │   └── 161294 repo
│       │   │       ├── 138372 src
│       │   │       │   ├── 119320 core
│       │   │       │   │   ├── 66129 thread
│       │   │       │   │   │   ├── 14470 mle_router.cpp
│       │   │       │   │   │   ├── 13047 mle.cpp
│       │   │       │   │   │   ├── 4602 network_data_leader_ftd.cpp
│       │   │       │   │   │   ├── 4602 lowpan.cpp
│       │   │       │   │   │   ├── 3882 mesh_forwarder.cpp
│       │   │       │   │   │   ├── 3420 address_resolver.cpp
│       │   │       │   │   │   ├── 3150 network_diagnostic.cpp
│       │   │       │   │   │   ├── 3090 mesh_forwarder_ftd.cpp
│       │   │       │   │   │   └── 2166 network_data.cpp
│       │   │       │   │   ├── 15328 meshcop
│       │   │       │   │   │   └── 3154 dataset_manager.cpp
│       │   │       │   │   ├── 10535 mac
│       │   │       │   │   │   ├── 4641 mac.cpp
│       │   │       │   │   │   └── 2180 mac_frame.cpp
│       │   │       │   │   ├── 9672 net
│       │   │       │   │   │   └── 3144 ip6.cpp
│       │   │       │   │   ├── 5562 common
│       │   │       │   │   ├── 4818 coap
│       │   │       │   │   │   └── 3080 coap.cpp
│       │   │       │   │   ├── 2700 utils
│       │   │       │   │   └── 2654 api
│       │   │       │   └── 19052 cli
│       │   │       │       ├── 13136 cli.cpp
│       │   │       │       └── 4216 cli_dataset.cpp
│       │   │       ├── 18022 third_party
│       │   │       │   └── 18022 NordicSemiconductor
│       │   │       │       └── 18022 drivers
│       │   │       │           └── 18022 radio
│       │   │       │               ├── 6972 nrf_802154_core.c
│       │   │       │               │   └── 2251 nrf_802154_core_irq_handler
│       │   │       │               ├── 3152 mac_features
│       │   │       │               └── 2300 rsch
│       │   │       └── 4900 examples
│       │   │           └── 4900 platforms
│       │   │               └── 4900 nrf528xx
│       │   │                   └── 4900 src
│       │   └── 20208 lwip
│       │       └── 20208 repo
│       │           └── 20208 lwip
│       │               └── 20208 src
│       │                   └── 19764 core
│       │                       ├── 4428 tcp_in.c
│       │                       │   └── 2748 tcp_input
│       │                       ├── 3764 tcp_out.c
│       │                       ├── 3080 tcp.c
│       │                       └── 2868 ipv6
│       ├── 49956 src
│       │   ├── 10318 transport
│       │   │   └── 2112 RendezvousSession.cpp
│       │   ├── 10188 platform
│       │   │   └── 8898 nRF5
│       │   │       ├── 3084 ThreadStackManagerImpl.cpp
│       │   │       └── 2968 BLEManagerImpl.cpp
│       │   ├── 9086 app
│       │   │   └── 8204 util
│       │   │       └── 2084 util.cpp
│       │   ├── 6680 ble
│       │   │   └── 3772 BLEEndPoint.cpp
│       │   ├── 3578 crypto
│       │   ├── 2884 inet
│       │   ├── 2510 lib
│       │   ├── 2276 system
│       │   └── 2112 setup_payload
│       └── 2816 examples
├── 85594 [section .text]
│   └── 2620 mbedtls_ecp_mul_restartable
├── 46320 home
│   └── 46320 kasr
│       └── 46320 repos
│           └── 46320 ncs
│               └── 44580 mbedtls
│                   └── 44580 library
│                       ├── 12472 ssl_tls.c
│                       │   └── 2588 mbedtls_ssl_read_record
│                       ├── 6168 bignum.c
│                       ├── 4740 ssl_srv.c
│                       │   └── 4447 mbedtls_ssl_handshake_server_step
│                       ├── 4046 ssl_cli.c
│                       │   └── 3903 mbedtls_ssl_handshake_client_step
│                       ├── 3110 x509_crt.c
│                       └── 2464 ecjpake.c
├── 34710 [section .text]
└── 29514 nrf5sdk
    ├── 15386 components
    │   ├── 11506 libraries
    │   │   ├── 3508 fds
    │   │   │   └── 3508 fds.c
    │   │   └── 2634 log
    │   │       └── 2634 src
    │   └── 2528 ble
    ├── 9040 external
    │   └── 7642 freertos
    │       └── 6558 source
    │           ├── 3216 tasks.c
    │           └── 2106 queue.c
    └── 4040 modules
        └── 4040 nrfx
            └── 3508 drivers
                └── 3508 src
```
One item of interest is 34710 bytes that don't belong to any symbol.

kpschoedel, 2020-10-28:
Filed #3504 — Audit string constants

dougleaf, 2021-06-17:
Hi @kpschoedel , I'm also interested on tracking memory utilization. I typically use nm, but that gives us a long flat file list. How did you generate this graph?
@AliakbarEski

kpschoedel, 2021-06-17:
> Hi @kpschoedel , I'm also interested on tracking memory utilization. I typically use nm, but that gives us a long flat file list. How did you generate this graph?

The current version of the script is `scripts/tools/memory/report_tree.py`. See the [README](https://github.com/project-chip/connectedhomeip/blob/master/scripts/tools/memory/README.md) and/or `report_tree.py --help` to get started.

kpschoedel, 2022-01-31:
As of 8b503ad05993ccef1063c1b38d9a906a85955445, usage of RAM (i.e. sections marked writable, typically .data and .bss) across builds tracked in CI is:

| platform   | config                      | target                 |    size |
|------------|-----------------------------|------------------------|---------|
| cyw30739   | CYW30739                    | lock-app               |  535402 |
|            | cyw930739m2evb_01           | light                  |  577386 |
| efr32      | BRD4161A                    | lighting-app           |  125952 |
|            |                             | window-app             |  124580 |
|            | BRD4161A+rpc                | lighting-app           |  142608 |
| esp32      | c3devkit                    | all-clusters-app       | 1398882 |
|            | m5stack                     | all-clusters-app       |  463244 |
| k32w       | k32w061+release             | light                  |  662060 |
|            |                             | lock                   |  663120 |
| linux      | arm64                       | chip-tool-ipv6only     |  284689 |
|            |                             | thermostat-no-ble      |  149441 |
|            | debug                       | all-clusters-app       |  146408 |
|            |                             | chip-tool              |  181464 |
|            |                             | door-lock-app          |  118712 |
|            |                             | ota-provider-app       |   88480 |
|            |                             | ota-requestor-app      |   91512 |
|            |                             | shell                  |  148184 |
|            |                             | tv-app                 |  321304 |
|            | debug+rpc                   | bridge-app             |   88832 |
|            |                             | lighting-app           |  123672 |
| mbed       | CY8CPROTO_062_4343W+release | all-clusters-app       | 2382928 |
|            |                             | lighting-app           | 2337496 |
|            |                             | lock-app               | 2301600 |
|            |                             | pigweed-app            | 1139712 |
|            |                             | shell                  | 2292204 |
| nrfconnect | nrf52840dk_nrf52840         | lighting-app           |  980759 |
|            |                             | lock-app               |  913339 |
|            |                             | pigweed-app            |  541779 |
|            |                             | pump-app               |  916195 |
|            |                             | pump-controller-app    |  911387 |
|            |                             | shell                  |  798379 |
|            | nrf52840dk_nrf52840+rpc     | lighting-app           |  966303 |
|            | nrf52840dongle_nrf52840     | lighting-app           |  996827 |
|            | nrf5340dk_nrf5340_cpuapp    | lighting-app           |  889926 |
|            |                             | lock-app               |  823402 |
| p6         | default                     | all-clusters-app       | 2443728 |
|            |                             | light-app              | 2339240 |
|            |                             | lock-app               | 2304712 |
| qpg        | qpg6105+debug               | lighting-app           |  146940 |
|            |                             | lock-app               |  146940 |
|            |                             | persistent-storage-app |  146940 |
| telink     | tlsr9518adk80d              | lighting-app           |  845182 |

Smallest is Linux ota-provider-app at 88480 bytes.

kpschoedel, 2022-01-31:
RAM usage is labelled ‘(read/write)’ in PR size reports; e.g. the most recent green run at the moment is https://github.com/project-chip/connectedhomeip/pull/14310#issuecomment-1025174239

turon, 2022-01-31:
We have the measurements, but don't want to close until there is a follow-up issue to analyze the data and track shrinking the sizes.  Cursory analysis:
- RAM sizes are generally large for most examples on most platforms.
- We are very close to exceeding our target of <128kb RAM, even on simple examples.
- Some seemingly similar platforms have vastly different sizes.
   Example - Cortex-M platforms are either at expected sizes (efr32, qpg) or vastly over (nrfconnect, k32w, mbed).
   This may be explained somewhat by the fact that platforms that mark `.text` as writable, get that counted as RAM.
   But follow-up required to build reports that are more readily useful for assessing where we are at versus our goals.

Removing `v1_triage_split_2`.

closed 2023-07-11 by stale[bot]