project-chip/connectedhomeip #2824: Sort out whether key ids are 16-bit or 32-bit

bzbarsky-apple, 2020-09-25:
#### Problem
Key ids seem to be 16-bit in `SecurePairingSession` and in our packet headers.  But they're 32-bit in `PeerConnectionState`.  This causes `-Wconversion` warnings when key IDs from `PeerConnectionState` move into one of the other places... and those warnings are correct, if the key ids can be out of the 16-bit value range!

 #### Proposed Solution
Decide whether key IDs are 16-bit or 32-bit and be consistent about it.

@pan-apple do you know whether there's a reason for `PeerConnectionState` to not use `uint16_t` for key ids?

pan-apple, 2020-09-25:
@bzbarsky-apple, the encryption key ID is spec'ed out to be 16 bits. We can change `PeerConnectionState` to use `uint16_t`.

bzbarsky-apple, 2020-09-25:
Fixed by https://github.com/project-chip/connectedhomeip/pull/2810

closed 2020-09-28 by BroderickCarlin