project-chip/connectedhomeip #2855: Codesize measurements seem to not be happening

bzbarsky-apple, 2020-09-26:
#### Problem
Codesize measurements are not happening on PRs.  For example, https://github.com/project-chip/connectedhomeip/pull/2759 was a big codesize win, but there were no change reports in that  from bloaty.

 #### Proposed Solution
Figure out why the automation is not running and fix it.

bzbarsky-apple, 2020-09-26:
@woody-apple

bzbarsky-apple, 2020-09-26:
https://github.com/project-chip/connectedhomeip/runs/1168479021?check_suite_focus=true says:
```
Run scripts/helpers/bloat_check.py                        \
Traceback (most recent call last):
  File "scripts/helpers/bloat_check.py", line 347, in <module>
    main()
  File "scripts/helpers/bloat_check.py", line 250, in main
    artifacts = [a for a in github_fetch_artifacts.getAllArtifacts(args.github_api_token, args.github_repository)]
  File "/__w/connectedhomeip/connectedhomeip/scripts/helpers/github_fetch_artifacts.py", line 91, in getAllArtifacts
    repo = api.get_repo(githubRepo)
  File "/usr/local/lib/python3.8/dist-packages/github/MainClass.py", line 333, in get_repo
    headers, data = self.__requester.requestJsonAndCheck(
  File "/usr/local/lib/python3.8/dist-packages/github/Requester.py", line 317, in requestJsonAndCheck
    return self.__check(
  File "/usr/local/lib/python3.8/dist-packages/github/Requester.py", line 342, in __check
    raise self.__createException(status, responseHeaders, output)
github.GithubException.RateLimitExceededException: 403 {"message": "API rate limit exceeded for installation ID 7125406.", "documentation_url": "https://docs.github.com/rest/overview/resources-in-the-rest-api#rate-limiting"}
```
which every single bloat check run looks like that....

woody-apple, 2020-10-04:
I tweaked one thing... curious if that helps

bzbarsky-apple, 2020-10-05:
Now looking at the list at https://github.com/project-chip/connectedhomeip/actions?query=workflow%3A%22Bloat+Check%22 I see some things with red "x" and some things with green checks.

A typical red "x" log looks like the thing I pasted above.  A typical "green check" log shows some fetching of artifacts, then something like:
```
 2020-10-05 04:25:12 ba5ad0741cde root[109] INFO Uploading report to "project-chip/connectedhomeip", PR 3044
2020-10-05 04:25:12 ba5ad0741cde root[109] WARNING Failed to process bloat report: AttributeError("'list' object has no attribute 'len'")
```
a number of times, then maybe some:
```
2020-10-05 04:48:55 e2162ded50f8 root[110] WARNING Failed to process bloat report: RateLimitExceededException(403, {'message': 'API rate limit exceeded for installation ID 7125406.', 'documentation_url': 'https://docs.github.com/rest/overview/resources-in-the-rest-api#rate-limiting'})
```
kind of things.

Well, that `len` thing is certainly something we should fix.  Seems like fallout from #2378.

I created https://github.com/project-chip/connectedhomeip/pull/3048 to fix the `len` issue and then we will see where things stand.

andy31415, 2020-10-05:
the len() fix should make things work again. Rate API limiting is annoying however the bloat report should be able to run ocassionally which is goood enough for us.

andy31415, 2020-10-05:
Reassigned to Boris since I believe #3048 is the real fix.

closed 2020-10-05 by rwalker-apple