project-chip/connectedhomeip #2862: Implement OTA Software Update cluster (Server side)

BroderickCarlin, 2020-09-28:
#### Problem
The CHIP spec defines an OTA software update cluster but the SDK currently does not implement said cluster's server side logic

 #### Proposed Solution
Implement the CHIP OTA software update cluster's server side logic

holbrookt, 2021-06-01:
@BroderickCarlin  - what is enough to constitute implementing the server side logic? #6476 implements the generic code for OTA Server logic, is this enough to close this issue?

BroderickCarlin, 2021-06-02:
@holbrookt - I would believe that #6476 fulfills the original intent of this issue. As pointed out in that PR we must rely on a vendor to implement a majority of their own software update system as it is inherently going to be highly vendor specific. The delegate approach, with some fleshed out boilerplate, should suffice to close this issue once merged.

PR #6476 (2021-06-10): OTAServerDelegate class for vendor logic
 #### Problem
Need a way to separate common code from application/vendor specific code for OTA Server Cluster, which doesn't require the developer to fork the repo.

#### Change overview
* define OTAServerDelegate class for implementing vendor OTA Server logic
* add some common logic for OTA Server Cluster commands
* add OTA Server Cluster to all-clusters-app and chip-tool
* fix AnnounceOTAServer command direction (OTA Client client node -> OTA Client server node)
* Fixes #2862 - Implement OTA Software Update cluster (Server side)

#### Testing
Ran `chi [... cut]
PR #7537 (2021-06-16): [Fixed] OTAServerDelegate class for vendor logic
This is a resubmit of #6476 with fixes for the breakages that lead to #7534

#### Problem
Need a way to separate common code from application/vendor specific code for OTA Server Cluster, which doesn't require the developer to fork the repo.

#### Change overview
* define OTAServerDelegate class for implementing vendor OTA Server logic
* add some common logic for OTA Server Cluster commands
* add OTA Server Cluster to all-clusters-app and chip-tool
* fix AnnounceOTAServer command direction (OTA Client client node -> OTA Client server node)
* fixe [... cut]
closed 2021-06-10 by woody-apple