project-chip/connectedhomeip #2877: Update Basic Cluster to align with CHIP spec

BroderickCarlin, 2020-09-28:
#### Problem
The currently implemented Basic Cluster is pulled unmodified from the ZCL spec and does not align with the CHIP defined Basic Cluster.

 #### Proposed Solution
Update the Basic Cluster implementation to align with the CHIP Basic Cluster

kpschoedel, 2021-02-25:
@vivien-apple — I have XML changes for this in #5019

bzbarsky-apple, 2021-03-05:
Note: #5019 was reverted.

kpschoedel, 2021-03-05:
I brought this up in #zap earlier — it's associated with two .zap json records that are not significantly different from all the others, and goes away if the `code` is not either 5 or 6. It doesn't appear to affect the generated code (although I may have missed something).

franck-apple, 2021-05-18:
This is done. Closing.

closed 2021-05-18 by franck-apple