project-chip/connectedhomeip #2880: Add CHIP support for the Power Configuration and Battery cluster to one of the demo projects

BroderickCarlin, 2020-09-28:
#### Problem
CHIP defines means for reading information about a device's battery and/or power source but does not currently use this functionality in any of the demo applications.

 #### Proposed Solution
Add CHIP support for the Power Configuration and Battery cluster to one of the demo projects

fessehaeve, 2021-11-17:
We already have power source cluster. Is anyone working on Power source configuration cluster right now?
Is there something else related to battery?

fessehaeve, 2021-11-19:
@woody-apple @franck-apple anyone working on this?

woody-apple, 2022-01-25:
@fessehaeve Nope, unfortunately not! :(

fessehaeve, 2022-01-25:
@woody-apple apparently I did the power source configuration cluster but forgot about this open issue. There is also a power source cluster in the example apps (all-cluster-apps and lock-app). The only thing missing is hooking up the power source to an actual implementation, eg for battery measurement.

yufengwangca, 2022-02-02:
It looks we just need to enable those cluster in one of example app zap and implement the relevant callback if needed. Maybe thermostat is a good candidate. 

@fessehaeve  please let me know if you need any help on this.

fessehaeve, 2022-02-02:
The clusters are enabled in all-clusters-app and lock-app. I was proposing having actual implementation with hardware integration. Not sure if that is really necessary. We could close the issue if everyone is satisfied.

woody-apple, 2022-02-07:
This looks complete, yes. Thanks!

closed 2022-02-07 by woody-apple