project-chip/connectedhomeip #2893: source ./scripts/activate.sh behind a proxy failed

butterl, 2020-09-28:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >
It seems when run `source ./scripts/activate.sh` behind would go timeout which trying to downloading CIPD with pigweed， check the doc and src， seems no proxy setting could be passed,  nor the http_proxy environ is read

 #### Proposed Solution
<suggested fix, suggested enhancement>
Add a proxy configure or just read proxy setting fron ENV

butterl, 2020-09-29:
From the traceback seems when downloading cipd, it stuck in GET.  
```
Traceback (most recent call last):
  File "/home/chip/code/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 452, in <module>
    sys.exit(main())
  File "/home/chip/code/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 448, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/chip/code/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 260, in setup
    result = step()
  File "/home/chip/code/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 311, in cipd
    cipd_client = cipd_wrapper.init(install_dir, silent=True)
  File "/home/chip/code/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 290, in init
    bootstrap(client, silent)
  File "/home/chip/code/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 254, in bootstrap
    tmp.write(client_bytes())
  File "/home/chip/code/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 196, in client_bytes
    conn.request('GET', path)
  File "/usr/lib/python3.8/http/client.py", line 1255, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1301, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1250, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1010, in _send_output
    self.send(msg)
  File "/usr/lib/python3.8/http/client.py", line 950, in send
    self.connect()
  File "/usr/lib/python3.8/http/client.py", line 1417, in connect
    super().connect()
  File "/usr/lib/python3.8/http/client.py", line 921, in connect
    self.sock = self._create_connection(
  File "/usr/lib/python3.8/socket.py", line 808, in create_connection
    raise err
  File "/usr/lib/python3.8/socket.py", line 796, in create_connection
    sock.connect(sa)
OSError: [Errno 101] Network is unreachable
```

when downloading cipd mannually ,the cipd itself works well with default system proxy in ENV

butterl, 2020-09-30:
It looks that in `pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py` the download CIPD function not make a proxy env read

simple workaround that may fix this, test with both py2.7 and py 3.8 ，it works well in my env behind the proxy
```
diff --git a/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py b/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py
index 787347a..eb21774 100755
--- a/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py
+++ b/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py
@@ -28,6 +28,7 @@ import platform
 import ssl
 import subprocess
 import sys
+import base64
 
 try:
     import httplib
@@ -144,6 +145,27 @@ def expected_hash():
     raise Exception('platform {} not in {}'.format(expected_plat,
                                                    DIGESTS_FILE))
 
+def https_connect_with_proxy(target_url):
+    """create HTTPSConnection with proxy support."""
+
+    proxy_env = os.environ['https_proxy']
+    if proxy_env != '':
+        url = urlparse.urlparse(proxy_env)
+        conn = httplib.HTTPSConnection(url.hostname, url.port)
+        headers = {}
+        if url.username and url.password:
+            auth = '%s:%s' % (url.username, url.password)
+            #py2 and py3 need different socket type
+            py_version = sys.version_info.major
+            if py_version >= 3:
+                headers['Proxy-Authorization'] = 'Basic ' + str(base64.b64encode(auth.encode()))\
+                                                                .replace("b'", "").replace("'", "")
+            else:
+                headers['Proxy-Authorization'] = 'Basic ' + base64.b64encode(auth)
+        conn.set_tunnel(target_url, 443, headers)
+    else:
+        conn = httplib.HTTPSConnection(target_url)
+    return conn
 
 def client_bytes():
     """Pull down the CIPD client and return it as a bytes object.
@@ -157,7 +179,7 @@ def client_bytes():
         version = ins.read().strip()
 
     try:
-        conn = httplib.HTTPSConnection(CIPD_HOST)
+        conn = https_connect_with_proxy(CIPD_HOST)
     except AttributeError:
         print('=' * 70)
         print('''
@@ -212,7 +234,7 @@ brew uninstall python && brew install python
             location = res.getheader('location')
             url = urlparse.urlparse(location)
             if url.netloc != conn.host:
-                conn = httplib.HTTPSConnection(url.netloc)
+                conn = https_connect_with_proxy(url.netloc)
             path = '{}?{}'.format(url.path, url.query)
 
         # Some kind of error in this response.

```

butterl, 2020-09-30:
patch submited in pigweed repo and wait to see any better solution to this
https://pigweed-review.googlesource.com/c/pigweed/pigweed/+/19181

butterl, 2020-10-13:
the upstream of pigweed have merged the patch, this issue could be closed once it synced to CHIP

butterl, 2020-11-10:
code is merged in 7c9a74498b7038e77c0b81893cf7d5d1afa8f7a4， issue could close @woody-apple

closed 2020-11-10 by butterl

bekencorp, 2021-12-10:
i still encounted this issue with same phenomenon, and the current repo had contained this fix. Do i have to set some http_proxy to use the active.sh ? i didn't configure any proxy.

butterl, 2021-12-10:
@bekencorp   
you could try export  https_proxy  or  HTTPS_PROXY with your proxy setting
```
 export https_proxy=http://proxy_ip:port
 or export https_proxy=http://username:password@proxy_ip:port
```

solstice-gao, 2023-12-06:
> @bekencorp 您可以尝试使用您的代理设置导出 https_proxy 或 HTTPS_PROXY
> 
> ```
>  export https_proxy=http://proxy_ip:port
>  or export https_proxy=http://username:password@proxy_ip:port
> ```

这个非常有效、This is very effective.