project-chip/connectedhomeip #2902: Rendezvous over BLE is not working in ESP32 wifi-echo example

pan-apple, 2020-09-28:
#### Problem
`RendezvousDeviceDelegate` is failing to initialize in ESP32 example app. The error is coming from `BLE::Init()` call. I added some debug logs and this is where it's failing

```
BLE Init: has descriminator 0, conn obj 0
BLE transport init returned 4000047
E (1281) rendezvous-devicedelegate: RendezvousDeviceDelegate Init failure: Error 4000047 (0x003D092F)
```

```
CHIP_ERROR BLE::Init(RendezvousSessionDelegate * delegate, const RendezvousParameters & params)
{
    CHIP_ERROR err      = CHIP_NO_ERROR;
    BleLayer * bleLayer = params.GetBleLayer();

    VerifyOrExit(mState == State::kNotReady, err = CHIP_ERROR_INCORRECT_STATE);
    VerifyOrExit(bleLayer, err = CHIP_ERROR_INCORRECT_STATE);

    mDelegate = delegate;

    mBleLayer                           = bleLayer;
    mBleLayer->mAppState                = reinterpret_cast<void *>(this);
    mBleLayer->OnChipBleConnectReceived = OnNewConnection;

    printf("BLE Init: has descriminator %d, conn obj %d\n", params.HasDiscriminator(), params.HasConnectionObject());
    if (params.HasDiscriminator())
    {
        err = DelegateConnection(params.GetDiscriminator());
        printf("DelegateConnection ret %d\n", err);
    }
    else if (params.HasConnectionObject())
    {
        err = InitInternal(params.GetConnectionObject());
        printf("InitInternal ret %d\n", err);
    }
    else
    {
        err = CHIP_ERROR_INVALID_ARGUMENT;
    }
```

So, params doesn't have discriminator and connection object. If I change the last else condition to return `CHIP_NO_ERROR`, the Rendezvous flow starts working. But, that doesn't seem like the correct solution.

 #### Proposed Solution
Analyze and fix the issue.

pan-apple, 2020-09-28:
CC: @vivien-apple

gjc13, 2020-09-29:
These lines have already been removed in this PR
https://github.com/project-chip/connectedhomeip/pull/2889/files#diff-4338d8cc5c43404ed1aac8a2c7e7240fL67

I also ran into this issue when trying to enable rendezvous on the nrf platform.

closed 2020-09-30 by andy31415