project-chip/connectedhomeip #2910: mbedTLS NewCertificateSigningRequest fills wrong CSR length

nileshvora-sls, 2020-09-29:
**Problem**
I was looking into creating csr for OpCSR for my ESP32 platform. However I was getting CSR length 0 when I call **P256Keypair::NewCertificateSigningRequest**. Then I looked into definition of it, it has written something like below.
```
    result = mbedtls_x509write_csr_pem(&csr, out_csr, length, CryptoRNG, nullptr);
    VerifyOrExit(result >= 0, error = CHIP_ERROR_INTERNAL);
    csr_length = static_cast<unsigned int>(result);
```

**Proposed Solution**
According to mbedtls API description **mbedtls_x509write_csr_pem** API return error code, not length. For length, we need to write something like below.
```
    result = mbedtls_x509write_csr_pem(&csr, out_csr, length, CryptoRNG, nullptr);
    VerifyOrExit(result >= 0, error = CHIP_ERROR_INTERNAL);
    csr_length = strlen((char *)out_csr);
```

nileshvora-sls, 2020-09-29:
CC @pan-apple @bzbarsky-apple

pan-apple, 2020-09-29:
Thanks @nileshvora-sls , you are right about the return code. I'll push a fix for it.
The code was originally written to use `mbedtls_x509write_csr()` API, which returns the length. But in some iteration of the PR it got changed to PEM format.

PR #2913 (2020-09-29): Fix CSR creation using mbedTLS
 #### Problem
Incorrect length for CSR generated using `mbedTLS` library

 #### Summary of Changes
The return value for `mbedtls_x509write_csr_pem` was being used to set the length, whereas this is 0, or error code. 

fixes #2910
closed 2020-09-29 by bzbarsky-apple