project-chip/connectedhomeip #2922: Various inet tests fail in a Linux container on Mac

bzbarsky-apple, 2020-09-29:
#### Problem
https://github.com/project-chip/connectedhomeip/pull/2725 disabled `TestInetLayerDNS` on Mac, but I get the same failure when running in a Linux Docker container on a Mac host, along with others.  Specifically, I followed the following steps to reproduce:
```
% git clone https://github.com/project-chip/connectedhomeip.git
% cd connectedhomeip
% git submodule update --init
% ./integrations/docker/images/chip-build-nrf-platform/run.sh -- bash ./gn_build.sh
```
and the output I get is:
<details>
<summary>Output:</summary>

```
[16/2611] ACTION //src/inet/tests:TestInetEndPoint_run(//build/toolchain/host:linux_x64_clang)
FAILED: linux_x64_clang/gen/src/inet/tests/TestInetEndPoint_run.pw_pystamp 
python ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root /Users/bzbarsky/linux-connectedhomeip/ --current-path /Users/bzbarsky/linux-connectedhomeip/src/inet/tests --default-toolchain=//third_party/pigweed/repo/pw_toolchain/dummy:dummy --current-toolchain=//build/toolchain/host:linux_x64_clang --touch /Users/bzbarsky/linux-connectedhomeip/out/debug/linux_x64_clang/gen/src/inet/tests/TestInetEndPoint_run.pw_pystamp --capture-output -- /Users/bzbarsky/linux-connectedhomeip/third_party/pigweed/repo/pw_unit_test/py/pw_unit_test/test_runner.py --runner ../../third_party/pigweed/repo/targets/host/run_test --test linux_x64_clang/tests/TestInetEndPoint
INF Test 1/1: [ RUN] TestInetEndPoint
ERR ../../third_party/pigweed/repo/targets/host/run_test exited with status 255
OUT [457]
'#0:','inet-endpoint'
'#2:','Setup                           ','PASSED'
'#3:','InetEndPoint::PreTest           ','PASSED'
CHIP node ready to service events; PID: 462; PPID: 457
CHIP:IN: Async DNS worker thread woke up.
CHIP:IN: Posting DNS completion event to CHIP thread.
    DNS name resolution complete: 172.217.11.36
    DNS name resolution complete: 127.0.0.1
CHIP:IN: Async DNS worker thread woke up.
CHIP:IN: Posting DNS completion event to CHIP thread.
    DNS name resolution return no addresses
'#3:','InetEndPoint::ResolveHostAddress','PASSED'
'#3:','InetEndPoint::TestParseHost     ','PASSED'
'#3:','InetEndPoint::TestInetError     ','PASSED'
    Interfaces:
     interface id: 0x1, interface name: lo, interface state: UP, no multicast, no broadcast addr
     interface id: 0x2, interface name: tunl0, interface state: DOWN, no multicast, no broadcast addr
     interface id: 0x3, interface name: ip6tnl0, interface state: DOWN, no multicast, no broadcast addr
     interface id: 0x84, interface name: eth0, interface state: UP, supports multicast, has broadcast addr
    Addresses:
     127.0.0.1/8, interface id: 0x1, interface name: lo, interface state: UP, no multicast, no broadcast addr
     172.17.0.2/16, interface id: 0x84, interface name: eth0, interface state: UP, supports multicast, has broadcast addr
'#3:','InetEndPoint::TestInetInterface ','PASSED'
../../src/inet/tests/TestInetEndPoint.cpp:346: assertion failed: "err == INET_ERROR_WRONG_ADDRESS_TYPE"
../../src/inet/tests/TestInetEndPoint.cpp:357: assertion failed: "(err == INET_NO_ERROR) || (err == System::MapErrorPOSIX(EPERM))"
../../src/inet/tests/TestInetEndPoint.cpp:364: assertion failed: "(didBind && (err == INET_NO_ERROR)) || (!didBind && (err == INET_ERROR_INCORRECT_STATE))"
../../src/inet/tests/TestInetEndPoint.cpp:371: assertion failed: "(didBind && didListen && (err == INET_NO_ERROR)) || (!didBind && (err == INET_ERROR_INCORRECT_STATE))"
../../src/inet/tests/TestInetEndPoint.cpp:394: assertion failed: "addr.Type() == kIPAddressType_IPv6"
../../src/inet/tests/TestInetEndPoint.cpp:410: assertion failed: "err == INET_ERROR_WRONG_ADDRESS_TYPE"
../../src/inet/tests/TestInetEndPoint.cpp:458: assertion failed: "err == INET_ERROR_WRONG_ADDRESS_TYPE"
../../src/inet/tests/TestInetEndPoint.cpp:461: assertion failed: "err == INET_ERROR_WRONG_ADDRESS_TYPE"
'#3:','InetEndPoint::TestInetEndPoint  ','FAILED'
CHIP:IN: Raw endpoint pool FULL
CHIP:IN: Raw endpoint pool FULL
CHIP:IN: UDP endpoint pool FULL
CHIP:IN: TCP endpoint pool FULL
CHIP:IN: TCP endpoint pool FULL
CHIP:CSL: Timer pool EMPTY
CHIP:CSL: Timer pool EMPTY
CHIP:IN: Async DNS worker thread woke up.
CHIP:IN: Async DNS worker thread exiting.
CHIP:IN: Async DNS worker thread woke up.
CHIP:IN: Async DNS worker thread exiting.
'#3:','InetEndPoint::TestEndPointLimit ','PASSED'
'#4:','Teardown                        ','PASSED'
'#6:','1','7'
'#7:','8','114'
INF Test 1/1: [FAIL] TestInetEndPoint
[17/2611] ACTION //src/lib/support/tests:TestCHIPArgParser_run(//build/toolchain/host:linux_x64_clang)
[18/2611] ACTION //src/lib/support/tests:TestErrorStr_run(//build/toolchain/host:linux_x64_clang)
[19/2611] ACTION //src/lib/support/tests:TestCHIPMem_run(//build/toolchain/host:linux_x64_clang)
[... 149 lines cut ...]
    138.68.19.10
    138.68.19.10
CHIP:IN: Posting DNS completion event to CHIP thread.
DNS resolution complete for www.nest.com: 3 results returned
    35.241.20.76
    35.241.20.76
    35.241.20.76
'#3:','TestDNSResolution:Simultaneous     ','FAILED'
CHIP:IN: Async DNS worker thread woke up.
CHIP:IN: Async DNS worker thread exiting.
CHIP:IN: Async DNS worker thread woke up.
CHIP:IN: Async DNS worker thread exiting.
'#6:','4','8'
'#7:','9','102'
INF Test 1/1: [FAIL] TestInetLayerDNS
ninja: build stopped: subcommand failed.

```

</details>

 #### Proposed Solution
Would be good to understand whether this is a bug in something I am doing, or in the way the container is set up, or in the tests...

@mspang @yunhanw-google

holbrookt, 2022-02-01:
@bzbarsky-apple does this need to be fixed for 1.0?

bzbarsky-apple, 2022-02-01:
I don't know.  It means we don't have test coverage from those tests on Mac, but how valuable the tests are.... no idea.