project-chip/connectedhomeip #2942: [Darwin] Build failure because of the third_party/pigweed script

kkasperczyk-no, 2020-09-30:
#### Problem
I tried to build CHIP tool on iOS using Xcode, but I'm constantly getting an error brought by environment setup script for the third_party/pigweed. I didn't modify anything and I just imported project straightly from repository. I'm not sure if somebody else is having similar problem, but I would be grateful for any comments and hints how to fix it. See the full error log on the picture below:
[image]
 
#### Proposed Solution

bzbarsky-apple, 2020-10-01:
@kkasperczyk-no Did you run `bash ./scripts/bootstrap.sh` in the repository root?  If not, does doing that help?

sagar-apple, 2020-10-01:
> @kkasperczyk-no Did you run bash ./scripts/bootstrap.sh in the repository root? If not, does doing that help?

To add to that, I think you need to `source scripts/activate.sh` as well.

bzbarsky-apple, 2020-10-01:
So I just reproduced the problem by running the following steps:

1. Clone the repo.
1. `git submodule update --init`
1. Try to build in XCode by opening `src/darwin/Darwin.xcworkspace` and pressing the "run" button.

There was a bunch of cipd stuff going on for a while before the build failure, so it sure looks like it was _trying_ to bootstrap; it just seems to have failed somehow.

Manually running `bash ./scripts/bootstrap.sh` does fix the problem.  Manually running `activate.sh` is not needed.

It might be worth looking into why the bootstrapping that XCode tries to do apparently fails....

kkasperczyk-no, 2020-10-01:
@bzbarsky-apple @sagar-apple yeah in the meantime I solved that problem by running ./gn_build.sh, which setup python environment and the pigweed problem disappeared. In the end no big deal, but maybe it would be nice to place some information for the the people starting to work with this project (like me). I've seen some other people having problem with building this project, so I guess it's not so obvious.

bzbarsky-apple, 2020-10-01:
> In the end no big deal

I think it's a big deal; we should make it as simple as possible for people to build things.

I'm actually going to reopen this and do a bit of digging to see what's going on here.

bzbarsky-apple, 2020-10-01:
Comparing the state of the tree when the build failure occurs to the state after successfully running bootstrap, the differences seem to be:

1. A bunch of differences in `.pyc` files.
2. The broken case has various files missing:
```
Only in connectedhomeip/third_party/pigweed/repo/.environment: activate.sh
Only in connectedhomeip/third_party/pigweed/repo/.environment: config.json
Only in connectedhomeip/third_party/pigweed/repo/.environment/python3-env/bin: lm3s6965evb_qemu_unit_test_runner
Only in connectedhomeip/third_party/pigweed/repo/.environment/python3-env/bin: stm32f429i_disc1_detector
Only in connectedhomeip/third_party/pigweed/repo/.environment/python3-env/bin: stm32f429i_disc1_test_client
Only in connectedhomeip/third_party/pigweed/repo/.environment/python3-env/bin: stm32f429i_disc1_test_server
Only in connectedhomeip/third_party/pigweed/repo/.environment/python3-env/bin: stm32f429i_disc1_unit_test_runner
Only in connectedhomeip/third_party/pigweed/repo/.environment/python3-env/bin: watchmedo
Only in connectedhomeip/third_party/pigweed/repo/.environment/python3-env/lib/python3.8/site-packages: _watchdog_fsevents.cpython-38-darwin.so
Only in connectedhomeip/third_party/pigweed/repo/.environment/python3-env/lib/python3.8/site-packages: lm3s6965evb-qemu-utils.egg-link
Only in connectedhomeip/third_party/pigweed/repo/.environment/python3-env/lib/python3.8/site-packages: pw-watch.egg-link
Only in connectedhomeip/third_party/pigweed/repo/.environment/python3-env/lib/python3.8/site-packages: stm32f429i-disc1-utils.egg-link
Only in connectedhomeip/third_party/pigweed/repo/.environment/python3-env/lib/python3.8/site-packages: watchdog
Only in connectedhomeip/third_party/pigweed/repo/.environment/python3-env/lib/python3.8/site-packages: watchdog-0.10.3-py3.8.egg-info
Only in connectedhomeip/third_party/pigweed/repo/.environment/python3-env/lib/python3.8/site-packages/wrapt: _wrappers.cpython-38-darwin.so
```
3. There are text differences in a few files:
```
third_party/pigweed/repo/.environment/python3-env/lib/python3.8/site-packages/easy-install.pth
third_party/pigweed/repo/.environment/python3-env/lib/python3.8/site-packages/wrapt-1.12.1-py3.8.egg-info/SOURCES.txt
third_party/pigweed/repo/.environment/python3-env/lib/python3.8/site-packages/wrapt-1.12.1-py3.8.egg-info/installed-files.txt
third_party/pigweed/repo/.environment/python3-env/pip-packages.log
third_party/pigweed/repo/.environment/python3-env/pip-requirements.log
````

It looks like the failing command is logging to `third_party/pigweed/repo/.environment/python3-env/pip-packages.log`.   I'll see if I can get a more useful diff from that one by excluding the timestamps at the beginning....

@mspang

bzbarsky-apple, 2020-10-01:
[Diff of the log files with timestamps stripped](https://github.com/project-chip/connectedhomeip/files/5313884/log3.txt)

This is not very informative to me so far.

bzbarsky-apple, 2020-10-01:
One other note: it looks like the XCode build stuff doesn't do a submodule init either; I get a different failure if I don't do that manually.  Maybe it should?

@sagar-apple @rwalker-apple

sagar-apple, 2020-10-01:
I thought it was reasonable to expect users to `source ./scripts/bootstrap.sh` once when they first clone the repo. I guess we could duplicate that in the XCode build stuff but Rob would know better.

rwalker-apple, 2020-10-01:
the xcode connector doesn't do bootstrap, but probably should

mspang, 2020-10-01:
Maybe dump the environment? And check if `pip install watchdog` works in the XCode environment?

rwalker-apple, 2020-10-01:
the use case is a fresh checkout, and the build connector skips a few things, like submodule update, that it probably shouldn't

rwalker-apple, 2020-10-01:
> Maybe dump the environment? And check if `pip install watchdog` works in the XCode environment?

The environment is dumped as part of the Xcode build.  I don't see why`pip install watchdog` decides that something has to land in "/private/var/folders/", that's not present in any environment variable that Xcode supplies to the connector...

mspang, 2020-10-01:
The earlier packages in the log also used /private/var/folders and those seemed to pass.

rwalker-apple, 2020-10-01:
> The earlier packages in the log also used /private/var/folders and those seemed to pass.

yeah, realizing that it looks as though the failure is for:
`/Users/rwalker/dev/connectedhomeip/third_party/pigweed/repo/.environment/python3-env/include/site/python3.8/watchdog`
or for:
`/private/var/folders/_z/ggss9rvd56j07gj39tf6f7k00000gn/T/pip-record-47ic9gab/install-record.txt`

rwalker-apple, 2020-10-01:
ok, the issue is that the Xcode environment is configured to build for iOS.  installation of watchdog tries to use the default compiler (`cc`) and fails because the environment is telling cc to build for iOS.

PR #2993 (2020-10-02): Update submodules in Xcode builds

 #### Problem
 Xcode builds fail on a freshly checked-out tree

 #### Summary of Changes
 * git submodule update --init
 * extra build output in case the build still fails, note that bootstrap.sh cannot be run in the Xcode iOS build environment

 fixes #2942
closed 2020-10-02 by bzbarsky-apple