project-chip/connectedhomeip #3089: Calculating key with SHA256 on nrfconnect and nrf5 platforms takes very long time.

kkasperczyk-no, 2020-10-06:
#### Problem
When initializing Rendezvous session there is a need to calculate key using pbkdf2_sha256() method. On the both nRF platforms - nrf5 and nrfconnect (as both are using mbedTLS library) it was observed that application is spending very long time on calculating this key, which is probably not acceptable. Time of calculations is strictly connected with the kSpake2p_Iteration_Count value, which stands for numbers of iterations performed in generating the key.

Observed conclusions:
* kSpake2p_Iteration_Count is constant value common for all platforms and equal to 50 000.
* Time spend on calculations is equal to over 50 s for 50 000 iterations at nrfconnect platform.
* Lowering kSpake2p_Iteration_Count value to 4096 leads to shortening calculation time to about 4,5 s, which is still far too long.
* Basing on analyzing configuration options it was checked that application is using hardware module for calculation, not the software one (which is desired and should fasten calculations). 

Questions to answer:
* What is the reason of spending such a long time on calculations?
* Do we need such a big number of iterations like 50 000? What is the background of selecting that value? It obviously increases the security but isn't this value far too big for the considered use case?

 #### Proposed Solution
Lowering the number of iterations may shorten the calculation time, but is not solving problem at all, so should be done only if the security requirements allow to this.

LuDuda, 2020-10-08:
@mspang @andy31415 @rwalker-apple could you look at this issue and comment on lowering the iteration number? We will still look at ways of improving calculation time.

rwalker-apple, 2020-10-09:
Would you mind providing a pointer to the code in question?

I can't imagine a case where a single sha256 wouldn't be equivalent to 50,000 iterations unless we're doing a VDF for something.

rwalker-apple, 2020-10-09:
Nevermind, I found a reference:

https://tools.ietf.org/html/rfc2898

```
4.2 Iteration Count

   An iteration count has traditionally served the purpose of increasing
   the cost of producing keys from a password, thereby also increasing
   the difficulty of attack. For the methods in this document, a minimum
   of 1000 iterations is recommended. This will increase the cost of
   exhaustive search for passwords significantly, without a noticeable
   impact in the cost of deriving individual keys.
```

IMHO: increasing the brute-force search space in a linear fashion isn't doing CHIP much good.  Widely available CPUs/GPUs can do 2M+ sha256s per second.  While the nrf board is grinding a key, there's an Intel or an AMD grinding at maybe 3 orders of magnitude faster.  The derived key is useful for a very short amount of time, in any case.

I think 1,000 iterations (or 1,024) should be sufficient for CHIP's purposes.

@balducci-apple, halp!

mspang, 2020-10-13:
This caused an issue for me too, opened a PR to lower.

Is 1000 low enough for nrf52480-dk ?

mspang, 2020-10-13:
https://github.com/project-chip/connectedhomeip/issues/2280 says we're going to pre-compute and persist this, which leads me to believe the current value can be set quite low because it is temporary.

rwalker-apple, 2020-10-13:
@pan-apple

pan-apple, 2020-10-13:
> @pan-apple 

I think @balducci-apple is looking into it.

PR #3205 (2020-10-13): Reduce kSpake2p_Iteration_Count to fix timeouts
Initializing this takes 15 seconds on ESP32, which is longer than most
timers we specify (e.g. BLE disconnect timeout is 5 seconds).

Reduce this to a value that doesn't interfere with other work.

fixes #3089

 #### Problem

Spake2p initialization causes timeouts.

 #### Summary of Changes

Reduce iterations by 100x.
commit 382008c7 (2020-10-13): Reduce kSpake2p_Iteration_Count to fix timeouts (#3205)

* Reduce kSpake2p_Iteration_Count to fix timeouts

Initializing this takes 15 seconds on ESP32, which is longer than most
timers we specify (e.g. BLE disconnect timer is 5 seconds).

Reduce this to a value that doesn't interfere with other work.

fixes #3089

* Update RendezvousSession.cpp

* Update RendezvousSession.cpp
closed 2020-10-13 by mspang