project-chip/connectedhomeip #3090: Manual Entry Setup Code ambiguity when Vendor ID and Product ID are 0

sagar-apple, 2020-10-06:
#### Problem
CHIP allows Vendor ID and Device IDs of 0000. 
With our current implementation this causes some ambiguity. 
For example, the short code `00000394906`and the long code `000003949100001000011` both return `0` for the Vendor ID and Product ID. 

However, the flag to indicate whether this accessory needs a "Custom Setup Flow" is set to `False` for the short code and `True` for the long code. They should ideally align.
 
 #### Proposed Solution
Two possible solutions.
1. Disallow VendorID and ProductIDs of `0000`. This has the downside of making it slightly harder to test with CHIP and also requires additional checks inside the CHIP SetupPayload object to intentionally return `Null` when these values are set to `0`.
2. Special case VID/PID `0000` such that the "Custom Setup Flow" flag is consistent.

sagar-apple, 2020-10-06:
@bhaskar-apple @vivien-apple @shana-apple FYI

gerickson, 2020-10-06:
A Vendor ID of 0x0000 should be allowed and should implicitly be thought of as "Common" or "Project CHIP Work Group". A Product ID of 0x0000 should be invalid.

andy31415, 2022-01-31:
Spec as of today says of PID:

```
A value of 0x0000 SHALL NOT be assigned to a product since Product ID = 0x0000 is used for these specific cases:
...
```

andy31415, 2022-01-31:
Closing as assuming this is "done" in the sense that product id 0 is not valid, so the described conflict should not be possible in spec-compliant implementations.

closed 2022-01-31 by andy31415