project-chip/connectedhomeip #3092: CHIPMem.h called from C

kpschoedel, 2020-10-06:
#### Problem

Part of #2748 All Embedded Device-side Code Needs to Be Scrubbed of Direct Heap Usage

Some C code currently calls malloc-family functions. Replacing these calls with CHIPMem.h
requires that it have an interface callable from C, ideally without adding overhead.

CHIPMem.h fundamentally provides a C-compatible interface, but wraps it in a namespace. The file also includes CHIPError.h, which is not currently C-compatible (again due to namespace).

 #### Proposed Solution

TBD

gerickson, 2020-10-06:
CHIP isn't / doesn't purport to be a C-compatible stack. Is there any need to provide C APIs for CHIPMem as a result?

kpschoedel, 2020-10-06:
If CHIP isn't intended to be C-compatible, then I suppose the burden can be on such platform C code to use a wrapper.

gerickson, 2020-10-06:
> If CHIP isn't intended to be C-compatible, then I suppose the burden can be on such platform C code to use a wrapper.

+1

PR #3143 (2020-10-09): Remove explicit malloc()-family calls from first-party code.
#### Problem

Embedded device-side code should only use heap through CHIPMem.h.
In particular it should not call malloc()-family functions.

#### Summary of Changes

- Replace direct calls to malloc()/calloc()/realloc()/free().
- Ensure that malloc()-family calls in non-device code are suitably
  guarded by #ifs.
- Add a C wrapper for Platform::Memory… functions, for
  src/lwip/standalone/sys_arch.c and platform C.
- Add a str[n]dup() equivalent, as calls are paired with free().

Fixes #2807 Remove explicit use of malloc() from first [... cut]
commit 41feafa5 (2020-10-09): Remove explicit malloc()-family calls from first-party code. (#3143)

* Remove explicit malloc()-family calls from first-party code.

#### Problem

Embedded device-side code should only use heap through CHIPMem.h.
In particular it should not call malloc()-family functions.

#### Summary of Changes

- Replace direct calls to malloc()/calloc()/realloc()/free().
- Ensure that malloc()-family calls in non-device code are suitably
  guarded by #ifs.
- Add a C wrapper for Platform::Memory… functions, for
  src/lwip/standalone/sys_arch.c and platform C.
- Add a str[n]dup() equivalent, as calls are pa [... cut]
closed 2020-10-09 by mspang