project-chip/connectedhomeip #3096: Merge common parts of the temperature measurement demo with other example apps

bzbarsky-apple, 2020-10-06:
#### Problem
Some code like `DataModelHandler` is the same in the temp sensor app and our lighting/lock examples.  Other code, like the overall server and rendezvous, is just similar/analogous.

 #### Proposed Solution
Converge `examples/common` and the temperature measurement demo app until the latter can use the code in `examples/common`

vivien-apple, 2020-11-02:
I did the `DataModelHandler` part in #3468. 
The other bits seems a duplicate of #2891 which is about the Server/Rendezvous bits.

vivien-apple, 2020-11-02:
Because of the previous comments I'm closing it. Feel free to reopen if I'm wrong ! :)

closed 2020-11-02 by vivien-apple

bzbarsky-apple, 2020-11-07:
No, it's ok.   There is stuff that's now compiled into the esp32 example but not used, but we can work on that....