project-chip/connectedhomeip #3104: Implement Secure Channel Protocol

yufengwangca, 2020-10-07:
#### Problem
Secure Channel Protocol defines the control plane for many facets of secure channel communication and security, including: message counter synchronization, reliable messaging acknowledgements.

 https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/secure_channel/Secure_Channel_Protocol.adoc

 #### Proposed Solution
Implement the Secure Channel Protocol as a CHIP protocol, please note, the session establish protocol(Sigma, Spake2+) is not within the scope of this implementation.

@woody-apple @pan-apple @andy31415 @robszewczyk

glerchundi, 2020-10-19:
I don't know which one would be the right forum to ask for this but, would it be possible to have access to the specifications repository?

yufengwangca, 2020-10-19:
> I don't know which one would be the right forum to ask for this but, would it be possible to have access to the specifications repository?

Maybe contact Chris LaPré first for access

glerchundi, 2020-10-20:
Thanks @yufengwangca I'll contact him then, thanks :D

woody-apple, 2020-10-22:
We're tracking high level tasks like this in projects: https://github.com/project-chip/connectedhomeip/projects/36

Please file issues there as needed!

closed 2020-10-22 by woody-apple