project-chip/connectedhomeip #3115: GetTickCounterFromTimeDelta returns uint32_t but we store its return value in uint16_t

bzbarsky-apple, 2020-10-07:
#### Problem
`ChipExchangeManager::GetTickCounterFromTimeDelta` has a long discussion about how it only overflows every so often, because it's 32-bit.  But we store its return value in 16-bit integers in various places like so:
```
mRMPNextAckTime   = ExchangeMgr->GetTickCounterFromTimeDelta(....)
...
mRMPThrottleTimeout = ExchangeMgr->GetTickCounterFromTimeDelta(...)
...
RetransTable[i].nextRetransTime = GetTickCounterFromTimeDelta(...)
```
If the full counter overflows in 46 days (per the comments), grabbing just the low 16 bits will overflow in about 1 minute.  This seems fairly suspect.

 #### Proposed Solution
Either change all those storage locations to `uint32_t` or if the 1-minute overflow is acceptable either change `GetTickCounterFromTimeDelta` to return `uint16_t` or add casts and comments explaining why the casts are OK at the spots where the size conversion happens.

@andy31415 do you happen to know what should happen here?

yufengwangca, 2020-10-28:
All of references of those code are within wml based lib/message, we have agreed to discard this stack and use new messaging stack from src/messaging and src/transport. This issue is not obsolete.

closed 2020-10-28 by yufengwangca

bzbarsky-apple, 2020-10-28:
@yufengwangca is there an issue tracking deleting the code we plan to discard?

yufengwangca, 2020-10-28:
> @yufengwangca is there an issue tracking deleting the code we plan to discard?

Yes, PR https://github.com/project-chip/connectedhomeip/pull/3445 remove the old stack from BUILD.gn

bzbarsky-apple, 2020-10-28:
Right, but is something tracking removing the no-longer-compiled code from the tree?

yufengwangca, 2020-10-28:
We would like to keep those code for reference until we cherry picked all needed features to current running stack.

https://github.com/project-chip/connectedhomeip/labels/message_layer_integration

Track the remaining issues of the integration