project-chip/connectedhomeip #3120: [android/CHIPTool] Deadlock after pushing On/Off/Toggle button

Damian-Nordic, 2020-10-07:
#### Problem
As a result of #2742 the CHIPTool for Android freezes when trying to push any button on the On/Off Cluster screen. The reason is that both CHIPDeviceController::beginConnectDevice and CHIPDeviceController::beginSendCommand take the same mutex while the second function is called in a callback of the first function.

 #### Proposed Solution
Bring a recursive mutex back?

@mspang Would you mind if I change the mutex to a recursive mutex again?

mspang, 2020-10-07:
Can we release the mutex while calling the callback?

mspang, 2020-10-07:
We can certainly revert the change to unblock but I'd still want to followup to fix the deadlock & switch back to a standard mutex.

To fix it, I believe we just need to unlock  while calling Java.

Damian-Nordic, 2020-10-07:
Yeah, there are actually 3 callbacks:
```
err = deviceController->ConnectDeviceWithoutSecurePairing(kRemoteDeviceId, deviceIPAddr, (void *) "ConnectDevice",
                                                              HandleKeyExchange, HandleEchoResponse, HandleError, CHIP_PORT);
```
All HandleKeyExchange, HandleEchoResponse and HandleError may actually want to do something with CHIP BLE stack, so all of them would need to implement the temporary mutex release which is doable, but quite error-prone as it assumes knowledge of when the callbacks are called. Why exactly using a recursive mutex is a problem?

mspang, 2020-10-07:
The problem with recursive locks is that it makes broken locking schemes fail less often.

Holding locks over callbacks is a bad design, now we have to think about whether Java ever calls the JNI code with another lock held, and whether that lock could also be taken in Java code that runs in the callback. A correct program needs to avoid a scenario where two locks are taken in opposite order on different threads.

It is a feature that with non-recursive locks we can identify these reentrancy cases because they deadlock in easily reproducible conditions.

PR #3128 (2020-10-08): [android] Fix deadlock in connectWithoutSecurePairing scenario
 #### Problem
As a result of #2742 the CHIPTool for Android freezes when trying to push any button on the On/Off Cluster screen. The reason is that both CHIPDeviceController::beginConnectDevice and CHIPDeviceController::beginSendCommand take the same mutex while the second function is called in a callback of the first function.

 #### Summary of Changes
Release the CHIP BLE stack mutex before invoking the successful connection callback.

 Fixes #3120
closed 2020-10-08 by BroderickCarlin