project-chip/connectedhomeip #3121: [Secure Channel] Support mDNS discovery for operational address resolution

balducci-apple, 2020-10-07:
#### Problem

CHIP accessories within a fabric need to determine the addresses of other accessories they wish to communicate with.

 #### Proposed Solution

Utilize mDNS to discover addresses. The mDNS query should include the fabric id and the node id of the intended accessory.

pan-apple, 2020-10-26:
We are currently noticing some build breakage due to missing mDNS for Darwin platform. There's a workaround to fix the build. But, eventually we'll need mDNS support for all platforms.

Damian-Nordic, 2021-04-13:
Note that we have already implemented the `Resolver` interface based on Linux (avahi), Darwin. Also, the minimal mDNS  implementation is pending (PR created by Andy: https://github.com/project-chip/connectedhomeip/pull/5907). The interface is then used in the DeviceController to update IP addresses in the internal structures (see https://github.com/project-chip/connectedhomeip/commit/e047d27d384e8243e19ccdcf85d5389eb45c10ac). Hence, I think the task may be too general and needs clarification.

franck-apple, 2021-04-15:
Controller to accessory is taken care of by #5956 / #5990.
Accessory to accessory remains to be done, but not needed in TE2
Modifying title and milestone to reflect this.

turon, 2022-01-31:
This is fixed -- SDK support mDNS lookup of operational IP addresses now.
Remove `v1_triage_split_4` tag.

closed 2022-01-31 by turon