project-chip/connectedhomeip #3164: wifi-echo build broken with CHIP_PROGRESS_LOGGING=1

msandstedt, 2020-10-08:
examples/wifi-echo/server/esp32 build is broken at HEAD of master at this hash:

```
commit 4be198bb1e02a9551e15a3ca93b8bad716e7f425 (HEAD -> master)
Author: Vivien Nicolas <vnicolas@apple.com>
Date:   Fri Sep 11 17:10:50 2020 +0200

    Break examples/chip-tool/main.cpp into multiple files (#2510)
```

It fails with:

```
connectedhomeip/src/inet/IPAddress.h:534:22: note:   no known conversion for argument 1 from 'const esp_ip4_addr_t' {aka 'const esp_ip4_addr'} to 'const ip4_addr_t&' {aka 'const ip4_addr&'}
src/platform/ESP32/ConnectivityManagerImpl.cpp:962:51: error: no matching function for call to 'chip::Inet::IPAddress::FromIPv4(const esp_ip4_addr_t&)'
         IPAddress::FromIPv4(got_ip.ip_info.netmask).ToString(netMaskStr, sizeof(netMaskStr));
```

The breakage can be overcome with this workaround:

```
diff --git a/src/lib/core/CHIPConfig.h b/src/lib/core/CHIPConfig.h
index 132b1b4d..b097426d 100644
--- a/src/lib/core/CHIPConfig.h
+++ b/src/lib/core/CHIPConfig.h
@@ -1798,7 +1798,7 @@
  *
  */
 #ifndef CHIP_PROGRESS_LOGGING
-#define CHIP_PROGRESS_LOGGING                              1
+#define CHIP_PROGRESS_LOGGING                              0
 #endif // CHIP_PROGRESS_LOGGING

 /**
```

vivien-apple, 2020-10-27:
@msandstedt Is it still broken for you ?

msandstedt, 2020-10-27:
> @msandstedt Is it still broken for you ?

No, this is now working at this commit, the current HEAD of master:

```
commit 960cfe0d29c364731d834b89a98f0a67e26aff2e (HEAD -> master, upstream/master)
Author: Damian Królik <66667989+Damian-Nordic@users.noreply.github.com>
Date:   Mon Oct 26 20:34:51 2020 +0100

    Handle Thread Association Request in rendezvous session (#3303)
```

I have not identified the commit that fixed this.

msandstedt, 2020-11-02:
Closing since this now works at hash 960cfe0d29c364731d834b89a98f0a67e26aff2e.

closed 2020-11-02 by msandstedt