project-chip/connectedhomeip #3171: nrfconnect examples does not work

erjiaqing, 2020-10-09:
#### Problem
nrfconnect example does not work

Steps:
1. Download the elf in artifact from CI
2. `arm-none-eabi-objcopy -O ihex chip-lighting.elf chip-lighting.hex`
3. `nrfjprog -f nrf52 --chiperase --program chip-lighting.hex`

The led on board does not flash as expected, and when pressing the buttons, it has no response.

When using gdb, 

```
(gdb) target remote localhost:2331
Remote debugging using localhost:2331
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x00071d4a in ?? ()
(gdb) symbol 'chip-lighting.elf' 
Reading symbols from chip-lighting.elf...
(gdb) bt
#0  0x00071d4a in printk (fmt=0x0 <sys_pm_notify_power_state_entry>)
    at /opt/nrfconnect/sdk-nrf/zephyr/lib/os/printk.c:456
#1  0x00000000 in ?? ()
(gdb) 
```

Seems this is not an useful backtrace.

 #### Proposed Solution
<suggested fix, suggested enhancement>

Damian-Nordic, 2020-10-09:
@erjiaqing Could you provide more details? nrfjprog version, which exactly binary failed to load? I took a random artifact from github, did the same steps (except that I also added "--reset" option to nrfjprog) and worked for me. 

Note that currently nRF52 examples take quite significant time to boot (30 seconds) due to #3089. Did you try to attach to the UART Console and see the logs?

erjiaqing, 2020-10-09:
Checked again, it does takes long time to boot, close this issue now.

closed 2020-10-09 by erjiaqing