project-chip/connectedhomeip #3181: Unable to build examples/lighting-app/nrfconnect

manuel-apple, 2020-10-09:
#### Problem
I am trying to build examples/lighting-app/nrfconnect on MacOS following the instructions on the README page / "Using Docker Container".

I am able to setup the container as instructed but after setup, I try to build:

west build -b nrf52840dk_52840

But I get the following error:

[8/545] cd /var/chip/examples/lighting-app/nrfconnect/build/modules/openthread/build && echo -e OPENTHREAD_CON...LAYED_TRX_ENABLED=1"\n"NRF_802154_ACK_TIMEOUT_ENABLED=1"\n"NRF52840_XXAA"\n"MBEDTLS_CONFIG_FILE="nrf-config.h
-e OPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS=0
OPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS_MANAGEMENT=0
OPENTHREAD_CONFIG_DHCP6_CLIENT_ENABLE=1
OPENTHREAD_CONFIG_DHCP6_SERVER_ENABLE=1
OPENTHREAD_CONFIG_DNS_CLIENT_ENABLE=1
OPENTHREAD_CONFIG_JOINER_ENABLE=1
OPENTHREAD_CONFIG_TMF_NETWORK_DIAG_MTD_ENABLE=1
OPENTHREAD_CONFIG_TMF_NETDATA_SERVICE_ENABLE=1
OPENTHREAD_CONFIG_IP6_SLAAC_ENABLE=1
OPENTHREAD_CONFIG_SNTP_CLIENT_ENABLE=1
PACKAGE_NAME=OPENTHREAD
PACKAGE_VERSION=g07f430dac
OPENTHREAD_CONFIG_THREAD_VERSION=OT_THREAD_VERSION_1_1
KERNEL
__ZEPHYR__=1
_FORTIFY_SOURCE=2
BUILD_VERSION=v2.3.0-rc1-ncs1-3626-g4d8217a9e3c9
__LINUX_ERRNO_EXTENSIONS__
USE_PARTITION_MANAGER=0
__PROGRAM_START
NRF_802154_IRQ_PRIORITY=1
NRF_802154_INTERNAL_RADIO_IRQ_HANDLING=0
NRF_802154_USE_RAW_API=1
NRF_802154_PENDING_SHORT_ADDRESSES=16
NRF_802154_PENDING_EXTENDED_ADDRESSES=16
NRF_802154_RX_BUFFERS=8
NRF_802154_CCA_MODE_DEFAULT=NRF_RADIO_CCA_MODE_ED
NRF_802154_CCA_CORR_LIMIT_DEFAULT=2
NRF_802154_CCA_CORR_THRESHOLD_DEFAULT=45
NRF_802154_CCA_ED_THRESHOLD_DEFAULT=45
NRF_802154_CSMA_CA_ENABLED=1
NRF_802154_TX_STARTED_NOTIFY_ENABLED=1
NRF_802154_FRAME_TIMESTAMP_ENABLED=1
NRF_802154_DELAYED_TRX_ENABLED=1
NRF_802154_ACK_TIMEOUT_ENABLED=1
NRF52840_XXAA
MBEDTLS_CONFIG_FILE=nrf-config.h
[10/545] Performing configure step for 'ChipConfig'
Done. Made 40 targets from 54 files in 828ms
[507/545] Performing build step for 'ChipLibBuild'
[96/161] c++ obj/third_party/connectedhomeip/src/lib/support/libSupportLayer.CHIPArgParser.cpp.o
FAILED: obj/third_party/connectedhomeip/src/lib/support/libSupportLayer.CHIPArgParser.cpp.o 
/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++ -MMD -MF obj/third_party/connectedhomeip/src/lib/support/libSupportLayer.CHIPArgParser.cpp.o.d -mthumb -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -Wall -Werror -Wno-deprecated-declarations -Wno-unknown-warning-option -Wno-missing-field-initializers -Wno-unused-but-set-variable -Wno-unused-variable -Wno-psabi -Wno-cast-function-type -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -std=gnu++11 -fno-rtti -Wno-non-virtual-dtor -Wno-deprecated-copy -isystem/var/ncs/nrf/drivers/mpsl/clock_control -isystem/var/ncs/zephyr/include -isystem/var/chip/examples/lighting-app/nrfconnect/build/zephyr/include/generated -isystem/var/ncs/zephyr/soc/arm/nordic_nrf/nrf52 -isystem/var/ncs/zephyr/lib/libc/newlib/include -isystem/var/ncs/zephyr/lib/fnmatch/. -isystem/var/ncs/zephyr/subsys/net/lib/sockets/. -isystem/var/ncs/zephyr/subsys/net/lib/openthread/platform/. -isystem/var/ncs/zephyr/subsys/settings/include -isystem/var/ncs/zephyr/subsys/bluetooth -isystem/var/ncs/nrf/include -isystem/var/ncs/nrf/lib/multithreading_lock/. -isystem/var/ncs/nrf/subsys/bluetooth/controller/. -isystem/var/ncs/nrfxlib/mpsl/include -isystem/var/ncs/nrfxlib/crypto/nrf_cc310_platform/include -isystem/var/ncs/nrfxlib/crypto/nrf_cc310_mbedcrypto/include/mbedtls -isystem/var/ncs/nrfxlib/crypto/nrf_cc310_mbedcrypto/include -isystem/var/chip/examples/lighting-app/nrfconnect/build/modules/nrfxlib/nrf_security/include -isystem/var/chip/examples/lighting-app/nrfconnect/build/modules/nrfxlib/nrf_security/include/mbedtls_generated/ -isystem/var/ncs/mbedtls/include -isystem/var/ncs/mbedtls/include/mbedtls -isystem/var/ncs/nrfxlib/softdevice_controller/include -isystem/var/ncs/modules/hal/cmsis/CMSIS/Core/Include -isystem/var/ncs/modules/hal/nordic/drivers/nrf_radio_802154/. -isystem/var/ncs/modules/hal/nordic/nrfx -isystem/var/ncs/modules/hal/nordic/nrfx/drivers/include -isystem/var/ncs/modules/hal/nordic/nrfx/mdk -isystem/var/ncs/modules/hal/nordic/. -isystem/var/ncs/modules/lib/openthread/zephyr/../include -isystem/var/ncs/modules/lib/openthread/zephyr/../examples/platforms -isystem/var/ncs/modules/debug/segger/rtt -isystem/var/ncs/modules/crypto/tinycrypt/lib/include -isystem/var/ncs/modules/lib/openthread/zephyr/../include -isystem/var/ncs/modules/lib/openthread/zephyr/../examples/platforms -DKERNEL -D__ZEPHYR__=1 -D_FORTIFY_SOURCE=2 -DBUILD_VERSION=v2.3.0-rc1-ncs1-3626-g4d8217a9e3c9 -D__LINUX_ERRNO_EXTENSIONS__ -DUSE_PARTITION_MANAGER=0 -D__PROGRAM_START -DNRF_802154_IRQ_PRIORITY=1 -DNRF_802154_INTERNAL_RADIO_IRQ_HANDLING=0 -DNRF_802154_USE_RAW_API=1 -DNRF_802154_PENDING_SHORT_ADDRESSES=16 -DNRF_802154_PENDING_EXTENDED_ADDRESSES=16 -DNRF_802154_RX_BUFFERS=8 -DNRF_802154_CCA_MODE_DEFAULT=NRF_RADIO_CCA_MODE_ED -DNRF_802154_CCA_CORR_LIMIT_DEFAULT=2 -DNRF_802154_CCA_CORR_THRESHOLD_DEFAULT=45 -DNRF_802154_CCA_ED_THRESHOLD_DEFAULT=45 -DNRF_802154_CSMA_CA_ENABLED=1 -DNRF_802154_TX_STARTED_NOTIFY_ENABLED=1 -DNRF_802154_FRAME_TIMESTAMP_ENABLED=1 -DNRF_802154_DELAYED_TRX_ENABLED=1 -DNRF_802154_ACK_TIMEOUT_ENABLED=1 -DNRF52840_XXAA -Os -fcheck-new -std=c++11 -Wno-register -fno-exceptions -fno-rtti -imacros /var/chip/examples/lighting-app/nrfconnect/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros /var/ncs/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wpointer-arith -Wno-address-of-packed-member -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/var/chip/examples/lighting-app/nrfconnect=CMAKE_SOURCE_DIR -fmacro-prefix-map=/var/ncs/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/var/ncs=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=nano.specs -D_SYS__PTHREADTYPES_H_ -DMBEDTLS_CONFIG_FILE=\<nrf-config.h\> -DMBEDTLS_PK_WRITE_C -DMBEDTLS_X509_CREATE_C -DMBEDTLS_X509_CSR_WRITE_C -isystem/var/ncs/zephyr/include/posix -isystem/var/ncs/zephyr/../mbedtls/include -I/var/chip/examples/lighting-app/nrfconnect -I/var/chip/examples/lighting-app/nrfconnect/main/include -DCHIP_SEPARATE_CONFIG_H=1 -I../../../../../config/nrfconnect/third_party/connectedhomeip/src/include -I../../../../../config/nrfconnect/third_party/connectedhomeip/src -Igen/include -I../../../../../config/nrfconnect/third_party/connectedhomeip/src/lib -I../../../../../config/nrfconnect/third_party/connectedhomeip/third_party/nlassert/repo/include -c ../../../../../config/nrfconnect/third_party/connectedhomeip/src/lib/support/CHIPArgParser.cpp -o obj/third_party/connectedhomeip/src/lib/support/libSupportLayer.CHIPArgParser.cpp.o
../../../../../config/nrfconnect/third_party/connectedhomeip/src/lib/support/CHIPArgParser.cpp: In function 'bool chip::ArgParser::IsShortOptionChar(int)':
../../../../../config/nrfconnect/third_party/connectedhomeip/src/lib/support/CHIPArgParser.cpp:83:12: error: 'isascii' was not declared in this scope
   83 |     return isascii(ch) && isgraph(ch);
      |            ^~~~~~~
../../../../../config/nrfconnect/third_party/connectedhomeip/src/lib/support/CHIPArgParser.cpp: In function 'bool chip::ArgParser::ParseArgsFromString(const char*, const char*, chip::ArgParser::OptionSet**, chip::ArgParser::NonOptionArgHandlerFunct, bool)':
../../../../../config/nrfconnect/third_party/connectedhomeip/src/lib/support/CHIPArgParser.cpp:496:18: error: 'strdup' was not declared in this scope; did you mean 'strcmp'?
  496 |     argStrCopy = strdup(argStr);
      |                  ^~~~~~
      |                  strcmp
../../../../../config/nrfconnect/third_party/connectedhomeip/src/lib/support/CHIPArgParser.cpp: In function 'void chip::ArgParser::PrintOptionHelp(chip::ArgParser::OptionSet**, FILE*)':
../../../../../config/nrfconnect/third_party/connectedhomeip/src/lib/support/CHIPArgParser.cpp:605:17: error: 'strcasecmp' was not declared in this scope; did you mean 'strncmp'?
  605 |             if (strcasecmp(helpGroupNames[nameIndex], optSets[optSetIndex]->HelpGroupName) == 0)
      |                 ^~~~~~~~~~
      |                 strncmp
../../../../../config/nrfconnect/third_party/connectedhomeip/src/lib/support/CHIPArgParser.cpp: In function 'bool chip::ArgParser::ParseBoolean(const char*, bool&)':
[... cut ...]
cc1plus: error: unrecognized command line option '-Wno-unknown-warning-option' [-Werror]
cc1plus: all warnings being treated as errors
[101/161] c++ obj/third_party/connectedhomeip/src/platform/libDeviceLayer.GeneralUtils.cpp.o
ninja: build stopped: subcommand failed.
FAILED: chip/src/ChipLibBuild-stamp/ChipLibBuild-build chip/lib/libCHIP.a 
cd /var/chip/examples/lighting-app/nrfconnect/build/chip && ninja
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /var/chip/examples/lighting-app/nrfconnect/build

LuDuda, 2020-10-09:
Hello Manuel,

I'm sorry for the inconvenience, it seems that the newest nRF Connect SDK/Zephyr brought some regression. We will look at it.

We have just filed PR with updated documentation that specifies the commit/tag of nRF Connect SDK that should be used for building CHIP applications and can be passed to the `setup` command (#3175).

For now, please switch to the recommended NCS version:

```
cd /var/ncs/nrf
git checkout 83764f
west update
cd /var/chip/examples/lighting-app/nrfconnect/
rm -rf build/
west build -b nrf52840dk_nrf52840
```

manuel-apple, 2020-10-09:
Thanks @LuDuda

manuel-apple, 2020-10-12:
@LuDuda 

I followed your instructions and was able to build it and flash it to a PCA10056. I see the LEDs flashing as expected and if I press button 1 for 6+ seconds all LEDs turn on, so the software seems to be working.

However, I am unable to see the  debug logs. I use JLINK_V686a on a Mac:
JLinkExe -device NRF52840_XXAA -if SWD -speed 4000 -autoconnect 1 ---> works properly
JLinkRTTClient ---> connects to localhost successfully

However, no logs are printed in the console. Any ideas? I have used earlier versions of JLINK but same result. 
I know  the setup  works as I can see the logs for other examples I used in the past.

manuel-apple, 2020-10-12:
Thanks @LuDuda. I was following the instructions on the CHIP repo for the lighting example. I am fine with  the UART. How can I access the logs from my Mac? It would be useful to update the README page so other people don't have the same issue.

LuDuda, 2020-10-12:
I agree @manuel-apple, that in general there is a lack of good documentation on how to test any CHIP accessory example in the repo. We plan to add such section soon, also to show how to perform Rendezvous (onboarding) using Mobile Phone.

To be honest, I'm not familiar with macOS, but I guess you can use simple tools like Putty, screen, minicom.
For example on Linux, i use screen: `$ sudo screen /dev/ttyACM0 115200`, though it is not the easiest tool to work with 😄

manuel-apple, 2020-10-12:
Thanks @LuDuda. I  tried the screen command (it is  the one  that  I  use regularly) but for  some reason I am not able to get any logs and I cannot submit any commands to the board. I will investigate

LuDuda, 2020-10-13:
@manuel-apple as you were able to read the UART logs, can we close the issue for now?

PR #3244 (2020-10-16): [nrfconnect] Fixed building issues that appeared after last NCS update.
 #### Problem
There are few building issues in the nrfconnect samples, which appeared after last NCS upmerge:
* No such file or directory when including "nrf_cc3xx_platform_mutex.h".
* 'isascii' was not declared in this scope in CHIPArgParses.cpp file.
* 'strcasecmp' was not declared in this scope in CHIPArgeParses.cpp file.
* Invalid conversion from 'const device*' to 'device*' in LightingManager.cpp file, when calling device_get_binding() method.

 #### Summary of Changes
* Temporarly added include path to the missing fi [... cut]
closed 2020-10-13 by manuel-apple