project-chip/connectedhomeip #3233: Use CHIPDevicePairingDelegate to prompt user for WiFi credentials in iOS CHIP tool app

pan-apple, 2020-10-13:
#### Problem
CHIPTool iOS application currently prompts the user to share the WiFi credentials (with device/accessory) when it receives Rendezvous message from the device. The `RendezvousSession` is currently bubbling up this message to the application layer.

`CHIPDevicePairingDelegate` callback mechanism was recently introduced to CHIP stack, using which `RendezvousSession` calls into upper layers to request WiFi network provisioning. The delegate is plumbed all the way to CHIPTool application, but the [function](https://github.com/project-chip/connectedhomeip/blob/382008c7cf8af44fc59fcb5f078757db62baa800/src/darwin/CHIPTool/CHIPTool/View%20Controllers/QRCode/QRCodeViewController.m#L288) is currently a no-op (it just prints a log message).

The application code needs to be changed to use this function to prompt the user for the network credentials, and send it to `RendezvousSession` using the included handler callback (passed as a parameter to the function). 

 #### Proposed Solution
Update the application code to use [this function](https://github.com/project-chip/connectedhomeip/blob/382008c7cf8af44fc59fcb5f078757db62baa800/src/darwin/CHIPTool/CHIPTool/View%20Controllers/QRCode/QRCodeViewController.m#L288) to prompt the user for the network credentials
Also, update [this function](https://github.com/project-chip/connectedhomeip/blob/382008c7cf8af44fc59fcb5f078757db62baa800/src/darwin/CHIPTool/CHIPTool/View%20Controllers/QRCode/QRCodeViewController.m#L270) to *not* prompt the user for the network credentials.

andy31415, 2020-10-21:
How does one make the CHIPTool on iOS receive a rendezvous message?

andy31415, 2020-10-21:
Apparently it was my ESP32 config that was not using BLE. once selecting BLE rendezvous, my iPod tries to connect (but fails with a 'no ble layer available' error that I still have to figure out).

PR #3378 (2020-10-22): ask and send credentials during the Rendezvous phase on iOS
 #### Problem
Need a Rendezvous implementation in iOS.

 #### Summary of Changes
Prompts the user for ssid/pwd and sends it to the device.

 Fixes #3233
closed 2020-10-22 by mspang