project-chip/connectedhomeip #3253: Message Security

turon, 2020-10-14:
#### Problem
Group security requires low level encryption/decryption pipeline to be implemented.

 #### Proposed Solution
Implement low levels of Message Security as defined in https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/241.

woody-apple, 2020-10-19:
Covered in:
https://github.com/project-chip/connectedhomeip/issues/1818
https://github.com/project-chip/connectedhomeip/issues/1819
https://github.com/project-chip/connectedhomeip/issues/1820
https://github.com/project-chip/connectedhomeip/issues/1821

turon, 2020-10-21:
Don't agree that implementation of encryption / decryption details is covered in those 4 "key management" issues.

woody-apple, 2020-11-05:
Can you please elaborate what is needed in addition specifically? We're trying to narrow down issues and clarify what's required.

closed 2020-11-05 by woody-apple