project-chip/connectedhomeip #3281: Building in XCode 12 for iOS simulator targets fails

bzbarsky-apple, 2020-10-16:
#### Problem
When I try to compile in XCode 12 for one of the iOS simulator targets (say iPhone 8), I get a link error like so:
```
ld: in /${HOME}/Library/Developer/Xcode/DerivedData/Darwin-gradnxuszkbudvafkyzpssssvewn/Build/Intermediates.noindex/CHIP.build/Debug iOS-iphonesimulator/CHIP.build/out/lib/libCHIP.a(libSetupPayload.SetupPayload.cpp.o), building for iOS Simulator, but linking in object file built for iOS, for architecture arm64
```

I checked older changesets, and before `b14e3e37d6038669cc46f849fc571b2f1e63c986` (switching XCode build to gn) the build would complete, but then XCode would complain that it couldn't run an arm64 binary in the simulator.

So it seems like going from XCode 11 to XCode 12 the simulator switched from running ARM binaries to x86_64 binaries?

 #### Proposed Solution
No idea offhand...

@rwalker-apple @bhaskar-apple @shana-apple @sagar-apple @andy31415

bzbarsky-apple, 2020-10-16:
> So it seems like going from XCode 11 to XCode 12 the simulator switched from running ARM binaries to x86_64 binaries?

Or perhaps we switched to compiling arm64 because `$ARCHS_STANDARD` changed?

andy31415, 2020-10-16:
I wressled with it and according to some google searches, it seems that historically simulators were x86_64 but arm macs may be coming so now the simulator seems to somehow crosscompile.

I got it working on my machine, but after a lot of trial and error (and unsure which of the fixes actually makes thins work). generally I tried to make everything just be x86_64 by:
  - updating valida ARCHS to include x86_64
  - add arm64 to excluded archs
  - did some manual add/remove frameworks and manual compiles and eventually it ran. Not convinced I could easily reproduce the success from scratch though

https://stackoverflow.com/questions/63607158/xcode-12-building-for-ios-simulator-but-linking-in-object-file-built-for-ios
https://developer.apple.com/forums/thread/656509

andy31415, 2020-10-16:
> > So it seems like going from XCode 11 to XCode 12 the simulator switched from running ARM binaries to x86_64 binaries?
> 
> Or perhaps we switched to compiling arm64 because `$ARCHS_STANDARD` changed?

This seems to be inline with what I read while trying to debug this issue - it now compiles both x64 and arm64 and our build setup does not like this.

bzbarsky-apple, 2020-10-16:
So for extra confusion, it seems like `VALID_ARCHS` is only ARM things for some reason?

rwalker-apple, 2020-10-16:
@andy31415 great work

I can look at this Monday

PR #3354 (2020-10-21): Fix busted Darwin builds and CHIPTool Simulator support
 #### Problem
The Darwin builds have regressed and don't seem to have support for XCode 12. 

1. The CHIPTool App fails compilation with a bitcode failure (the cpp code isn't being compiled with bitcode support)
2. The CHIPTool App crashes on launch on a device
3. Simulator builds don't work

 #### Summary of Changes
1. Disabled bitcode for the CHIP Framework to get it compiling again (fixes problem 1)
2. Added `DYLIB_INSTALL_NAME_BASE = @rpath` to the CHIP Framework build settings so that CHIPTool finds the framework installed correctly  [... cut]
closed 2020-10-21 by mspang