project-chip/connectedhomeip #3283: ESP32 builds should depend on sdkconfig.h

andy31415, 2020-10-16:
#### Problem
IDF build system generates sdkconfig.h from sdkconfig (built via `make menuconfig`)
GN-built files rely on the values in here (e.g. Platform initializes default SSID/PWD based on this file)
GN does not add a dependency on sdkconfig.h (-MMD option is passed when computing dependencies)

Generally the sdkconfig is a sideeffect include from:

```
In file included from /opt/espressif/esp-idf/components/newlib/platform_include/assert.h:20,
                 from /opt/espressif/esp-idf/components/esp_common/include/esp_err.h:18,
                 from /workspaces/connectedhomeip/examples/wifi-echo/server/esp32/third_party/connectedhomeip/src/platform/ESP32/CHIPPlatformConfig.h:28,
                 from /workspaces/connectedhomeip/examples/wifi-echo/server/esp32/third_party/connectedhomeip/src/system/SystemConfig.h:70,
                 from /workspaces/connectedhomeip/examples/wifi-echo/server/esp32/third_party/connectedhomeip/src/ble/BleConfig.h:42,
                 from /workspaces/connectedhomeip/examples/wifi-echo/server/esp32/third_party/connectedhomeip/src/lib/core/CHIPConfig.h:42,
                 from /workspaces/connectedhomeip/examples/wifi-echo/server/esp32/third_party/connectedhomeip/src/include/platform/CHIPDeviceConfig.h:31,
                 from /workspaces/connectedhomeip/examples/wifi-echo/server/esp32/third_party/connectedhomeip/src/include/platform/CHIPDeviceLayer.h:21,
                 from /workspaces/connectedhomeip/examples/wifi-echo/server/esp32/third_party/connectedhomeip/src/include/platform/internal/CHIPDeviceLayerInternal.h:22,
                 from ../../../../../../config/esp32/third_party/connectedhomeip/src/platform/ESP32/ConnectivityManagerImpl.cpp:22:
/workspaces/connectedhomeip/examples/wifi-echo/server/esp32/build/include/sdkconfig.h
```

Current understanding is that skdconfig.h is assumed to be a system include so it is not added as a dependency. When skdconfig is changed (`make menuconfig`) that will regenerate the header however libCHIP with gn is not rebuilt. This makes only clean builds work if sdkconfig is changed.

 #### Proposed Solution
Unsure ... building everything with GN may be ideal, but apart from that, any improvement would be good.

andy31415, 2020-10-16:
@mspang  found that

```
idf ninja -C build/chip all -d keepdepfile                      
cat ./build/chip/obj/third_party/connectedhomeip/src/platform/ESP32/libDeviceLayer.ConnectivityManagerImpl.cpp.o.d
```

shows the sdkconfig.h not being included and this is due to us passing all options from esp to ninja and this includes MMD (or at least this is my understanding)

MMD for deps seems wrong, since changing system includese clearly affects the build, so it does not make sense to exclude them.

mspang, 2020-10-16:
Rolling out `-isystem` is biting us badly here. It turns out this this drops all of those headers out from depfile generation (since everyone uses -MMD) and breaks incrementality.

@bzbarsky-apple

bzbarsky-apple, 2020-10-16:
So is the issue here that the `sdkconfig` file is in a path that's `-isystem` or that it's being included via things that are `-isystem`?  I would not have expected the former to be the case...

andy31415, 2020-10-16:
Everything under -isystem seems to be assumed as 'will not change' and do not trigger a gn rebuild.

however when I make menuconfig, the file changes and I would like for things to be rebuilt.

So far the pw suggestion is to manually add the dependency as a source, with a //-relative path.

mspang, 2020-10-16:
It's the former, all includes coming in from esp-idf have been tagged as system.

Regardless of the answer for sdkconfig, I also think we need to fix that updating dependencies also isn't going to cause rebuilds as appropriate.

bzbarsky-apple, 2020-10-16:
OK, I did some experimentation, and it's the latter for purposes of my earlier comment: if I have `-Ifoo -isystembar` and a header in `bar` includes something in `foo`, then the thing in `foo` does not end up in the `.d` file with `-MMD`.

Can we use `-MD` instead of `-MMD`?

mspang, 2020-10-16:
We can with a patch to pigweed, and I think I agree that we should. -MMD seems like a recipe for confusion.

andy31415, 2020-10-16:
We probably want:
 - pigweed patch (is this acceptable?)
 - patch fix_c_flags in config/esp32/components/chip/components.mk to also filter out -MMD

I made these two changes and now sdkconfig changes seems to also trigger a libCHIP recompile and the .d file looks to include system headers as well.

andy31415, 2020-10-16:
Apparently `#include <sdkconfig.h>` directly in CHIPPlatformConfig.h makes it seen as a dependecy since it is not included as a sideeffect.

This is totally not perfect, but it works around my SSID issue.

PR #3287 (2020-10-18): Add sdkconfig direct include in PlatformConfig, to depend outside -isystem

 #### Problem
sdkconfig.h is user editable (we generally update it with `make menuconfig` for things like wifi or other settings).

Currently this is included as a side-effect of including sys_err, however platform files use those defines. When performing an incremental build, gn assumes system includes never change and does not trigger a rebuild. This forces us to use a clean build every time sdkconfig is updated.

 #### Summary of Changes
Include sdkconfig.h directly instead of through system include sideffects.

Cha [... cut]
closed 2020-10-18 by mspang