project-chip/connectedhomeip #3286: Merge src/transport and src/messaging in github

yufengwangca, 2020-10-16:
#### Problem

Based on the latest messaging layer diagram:

Messaging layer is responsible for message framing & routing: With an interaction encrypted and signed, the Message Layer constructs the payload format with required and optional header fields; which specify properties of the message as well some routing information. In current CHIP codebase, ExchangeManager and SecureSessionMgr fall under this layer.

Transport layer is responsible for IP framing & transport management: After the final payload has been constructed, it is sent to the underlying transport protocol for IP management of the data. In current CHIP codebase, Inet falls under this layer.

 #### Proposed Solution
Merge the components under src/transport into src/messaging in github

woody-apple, 2020-10-21:
Per the architecture diagram, these are separate items. We shouldn't be merging base IP transport...

closed 2020-10-21 by woody-apple

yufengwangca, 2020-10-22:
@woody-apple your diagram show the SecureSessionMgr, SecureSession.... are under messaging.

It seems all components under following following directory should be merged to src/messaging.
https://github.com/project-chip/connectedhomeip/tree/master/src/transport

woody-apple, 2020-10-22:
Happy to sync up offline, but the TT diagram has a transport box which contains the base transport (TCP/UDP). 

There are many things to clean up here for sure, much of what's in transport needs to move around. BLE should move to rendezvous, rendezvous* should move to /rendezvous/ and secure session should move to likely into lib/message. 

We also need to be more explicit about the API that's allowed outside the messaging box (the Secure Channel API). @pan-apple can you help write up some documentation on that, and help us clean up headers here so it's clear what's private to the message implementation, vs usable outside of Secure Channel?

yufengwangca, 2020-10-22:
I think the transport box in TT diagram should also contains Inet which defines the TCP/UDP endpoints. I agree we should clean them up later with more specific tasks.

Thanks for clarification.