project-chip/connectedhomeip #3306: Align MessageHeader encoding/decoding with spec

yufengwangca, 2020-10-19:
#### Problem
Currently, MessageHeader is missing the logic to encode/decode ExchangeHeader flags. some flags are defined in the wrong place. Such as kVendorIdPresent should be defined in the ExchangeHeader instead of MessageHeader

 #### Proposed Solution
Align the message header encoding/decoding with the MessageFormat defined in the latest spec:  https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/secure_channel/Message_Format.adoc

woody-apple, 2020-10-22:
@yufengwangca  Can we break this into actionable updates so we can separate tasks?
eg: " as kVendorIdPresent should be defined in the ExchangeHeader instead of MessageHeader"

yufengwangca, 2020-11-19:
The current design re-use the flags, it has certain advantages than splitting it clearly out. Keep the current design for now until we see there is really need to refactor it.

closed 2020-11-19 by yufengwangca