project-chip/connectedhomeip #3308: Implement a simple protocol in CHIP for connectivity sanity test for messaging layer

yufengwangca, 2020-10-19:
#### Problem
By the nature of Messaging Layer, we need a basic protocol to sanity test the connectivity of the Messaging Layer on a higher level. In OpenWeave, this protocol is Echo. We will have echo feature implemented in cluster, but we also need a Echo protocol on protocol layer for basic connectivity test.

 #### Proposed Solution
Implement Echo protocol along side with Echo cluster for connectivity test for Messaging Layer. Then port existing test scripts from OpenWeave Happy/Cirque to CHIP for end to end test using this protocol.

@rwalker-apple @andy31415 @pan-apple 

@woody-apple please lets me know if you are ok with this plan or not.

pan-apple, 2020-10-19:
@yufengwangca , is the proposal to add echo protocol on top of exchange manager? Are there any requirements to send echo request/reply without encryption?

yufengwangca, 2020-10-19:
@pan-apple Yes, Echo protocol is implemented on top of Exchange Layer.

This feature does not mandate the encryption, but I notice we need to have a secure pairing before we can send/receive a message to peer in current stack, right? If that is the case, then we can make Echo protocol use encryption.

pan-apple, 2020-10-19:
Yes, it'll require the device to be paired with the controller (the two ends of the protocol).
Given that, how much value is it really bringing by adding this protocol? Echo cluster is just one layer above it (in ZCL protocol).

yufengwangca, 2020-10-19:
> Yes, it'll require the device to be paired with the controller (the two ends of the protocol).
> Given that, how much value is it really bringing by adding this protocol? Echo cluster is just one layer above it (in ZCL protocol).

To perform connectivity test on Message Layer only without involving ZCL.

rwalker-apple, 2020-10-19:
@pan-apple there isn't an echo cluster yet or an echo command on basic cluster.  Those could be added in order to verify that data model dispatch is intact and working.  An echo protocol alongside DM allows diagnostics on lower layers.

yes: everything will be above secure channel layer

closed 2020-11-04 by andy31415