project-chip/connectedhomeip #3343: Rendezvous Spake2p fails on EFR MG21 family

jmartinez-silabs, 2020-10-20:
#### Problem
Rendezvous Spake2p fails on EFR MG21 family. examples built and run on BRD4180A.

here is the Log error messages at bootup.
```
<error > [CR] mbedTLS error: 
<error > [SVR] ERROR setting up transport: Error 4000172 (0x003D09AC)
```

I tracked down to fault to 
```
gRendezvousServer.Init()
  RendezvousSession::WaitForPairing()
    SecurePairingSession::WaitForPairing()
      mSpake2p.ComputeL()
        mbedtls_ecp_mul()
          mbedtls_ecp_check_privkey(). fails with MBEDTLS_ERR_ECP_INVALID_KEY
```

mbedtls_ecp_mul() doesn't exist on MG21 (at least not accessible by the Cortex-M33).  It doesn't have a crypto accelerator architecture and it doesn't have acces to primitive ECP operations

 #### Proposed Solution
There is no abstraction for mbedtls work done in mSpake2p.ComputeL()

I don't know much about mbedtls but some people at silabs gave me the following suggestions

"You can calculate a public key from private key directly using SE Manager: sl_se_export_public_key()"
Also,
"It depends on what you want to use the key for.  If you are using it for ECDSA, I recommend mbedtls_ecdsa_genkey().  If you are using it for ECDH, I recommend mbedtls_ecdh_gen_public(). You can also try mbedtls_ecp_gen_key() or mbedtls_ecp_gen_keypair(). "

@mspang You worked on the Spake2p. Does that makes sense? What should be done here?

mspang, 2020-10-20:
Sounds like either an Mbed TLS configuration issue or a version update needed in EFR32 SDK.

jmartinez-silabs, 2020-10-21:
the functions  mbedtls_ecp_mul(), mbedtls_ecp_check_privkey() are not supported on this MCU familly since it doesn't have crypto acceleration. I. don't think its a configration issue, other mbedtls api need to be used for this family to calculate the key

woody-apple, 2021-01-29:
@jmartinez-silabs Is this still an issue?

jmartinez-silabs, 2021-01-29:
@woody-apple  No this issue was fixed. I will close this

closed 2021-01-29 by jmartinez-silabs