project-chip/connectedhomeip #3376: Message Synchronization Protocol

turon, 2020-10-21:
#### Problem
Group security requires Message Synchronization Protocol to be implemented.

#### Proposed Solution

Implement low levels of Message Security as defined in:
* CHIP-Specifications/connectedhomeip-spec#230.
* CHIP-Specifications/connectedhomeip-spec#256

woody-apple, 2021-01-29:
@kghost can this task be broken up at this point?

yufengwangca, 2021-02-03:
It seems #4626 is dup of this ticket, just synced with Mingjie, I will post my PR with this ticket instead

yufengwangca, 2021-02-08:
Reopen https://github.com/project-chip/connectedhomeip/issues/4626 which handle the protocol implementation, the Initiating Message Counter Synchronization part is left in this ticket after all dependencies are closed

closed 2021-03-03 by andy31415