project-chip/connectedhomeip #3383: validate Linux WiFi support on Pi

mspang, 2020-10-21:
CHIP builds on Pi now, but I've heard reports that CHIP is not playing nice with wpa_supplicant.

Make the CHIP WiFi connectivity support run smoothly on the Pi, and document any extra steps.

mspang, 2020-10-29:
Workaround for the reported wpa_supplicant issue:

`gdbus call -y -d fi.w1.wpa_supplicant1 -o /fi/w1/wpa_supplicant1 -m fi.w1.wpa_supplicant1.CreateInterface "{'Ifname': <'wlan0'>}"`

Run it before starting the sample app. Possibly CHIP should do this automatically at startup.

@yufengwangca

yufengwangca, 2020-10-30:
Tested  the Ubuntu server 64-bit, running the following command does not help: gdbus call -y -d fi.w1.wpa_supplicant1 -o /fi/w1/wpa_supplicant1 -m fi.w1.wpa_supplicant1.CreateInterface "{'Ifname': <'wlan0'>}"

This operation causes uncertain termination of wpa_supplicant and the device lost connectivity immediately. Until now, the 64-bit Desktop version seems still the only viable solution

closed 2020-11-03 by andy31415